grbl/grbl #1556: Motor Steppers not working

poplarbeargi, 2019-10-04:
hello there, i've encountered a problem, i'm using Arduino CNC Shield V3.00  with A4988 stepper motor drivers. The arduino got flashed with GRBL already and its connected. Wiring and power already done with 24V power supply, but despite all that, my stepper motors doesn't seem to work, help please.

here's the cnc shield: 
[image]

armature81, 2019-10-05:
Hi 
Check shield board fuse

brzez, 2019-10-12:
Hi,
I have a similar problem.

You could check if the motor would move with a test program:
```
# define EN 8 
# define X_DIR 5 
# define Y_DIR 6 
# define Z_DIR 7 
# define X_STP 2 
# define Y_STP 3 
# define Z_STP 4 

void step (boolean dir, byte dirPin, byte stepperPin, int steps)
{
	digitalWrite (dirPin, dir);
	delay (50);
	for (int i = 0; i <steps; i ++) {
		digitalWrite (stepperPin, HIGH);
		delayMicroseconds (1000);
		digitalWrite (stepperPin, LOW);
		delayMicroseconds (1000);
	}
}
void setup () {
	pinMode (X_DIR, OUTPUT); pinMode (X_STP, OUTPUT);
	pinMode (Y_DIR, OUTPUT); pinMode (Y_STP, OUTPUT);
	pinMode (Z_DIR, OUTPUT); pinMode (Z_STP, OUTPUT);
	pinMode (EN, OUTPUT);
	digitalWrite (EN, LOW);
}
void loop () {
	step (false, X_DIR, X_STP, 1000); // X axis motor reverse 1 ring, the 200 step is a circle.
	step (false, Y_DIR, Y_STP, 1000); // y axis motor reverse 1 ring, the 200 step is a circle.
	step (false, Z_DIR, Z_STP, 1000); // z axis motor reverse 1 ring, the 200 step is a circle.
	delay (1000);
	step (true, X_DIR, X_STP, 1000); // X axis motor forward 1 laps, the 200 step is a circle.
	step (true, Y_DIR, Y_STP, 1000); // y axis motor forward 1 laps, the 200 step is a circle.
	step (true, Z_DIR, Z_STP, 1000); // z axis motor forward 1 laps, the 200 step is a circle.
	delay (100);
}
```

In my case - the motor works with a test program, but doesn't with grbl.
In my case, when in grbl, the Z motor is actually the Y motor, and it only vibrates, when I send gcode to move.

You could check the motor wiring, the connector cables are sometimes wrong (eg. in my case).
I still didn't fix my issue, but the motor wiring was wrong, and I got the motor working using the arduino code. Still no luck with grbl.

chamnit, 2019-10-12:
Vibrating is an indication that Grbl is working and sending step commands to your motors. What's happening is that you are losing steps because you probably don't have your stepper driver current pots set appropriately (tiny circular knob on drivers) and/or your Grbl settings too aggressively configured (too high accelerations and/or max speeds).

brzez, 2019-10-12:
@chamnit 
When I use the stepper test program, the motor spins fine - I assumed, that the pots are set ok.
I'm using pure grbl, I didn't change anything in the config. I have assumed that it should basically work out of the box.

It is also weird, that when I connect the motor to X axis, grbl will recognize it as Y axis, so sending `Y100` will try to move the X motor