grbl/grbl #1557: Issue in sending Gcodes from one arduino to another

naveenneog, 2019-10-05:
I am trying to map joystick movements to G-Code. So sending 
`Serial.write("G91 G0 Y-0.01");` from one arduino to another in  serial way.
I have maintained the baudrate and connections properly still this is not working.

Any Idea on how this can be achieved would be very much helpful.

Thanks

MeJasonT, 2019-10-05:
try Serial.print, write is used for bytes ?
you may need to use a carriage return
ie Serial.println ()
grbl will also start up locked have you sent a command to unlock it before sending commands

naveenneog, 2019-10-09:
I was sending `G91 G0 X0.01` directly . Whats the startup command by the way? I skimmed through the documentation I didn't find the proper one.

MeJasonT, 2019-10-09:
$X - Kill alarm lock
The command set is located here
[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands](url)
Normal procedure is to home your machine on start up so it knows where it is before performing any other action - on completion the unlock is performed, does not always work 100% (on my builds anyways) so a manual unlock usually follows.  The next stage would be to set up/zero axis work coordinates for a job (the blank material you are going to use with your project).
Don't forget that your program can be greatly assisted by how the code is produced from your CAM postprocessor.  Im currently looking at automating some of my tool change functions - I can either have the postprocessor stop the spindle before a tool change or my application can perform tasks like stop the spindle when a tool change is expected.  I'm also using Machine coordinate system to move the machine to a park location which leaves my work coordinates unaffected - part XYZ zero remains unaffected so when the tool change is completed and the program is advanced the spindle goes back to where its supposed to start the next code block.

Just for reference we are using the old abandoned grbl forum, we really should be here
[https://github.com/gnea/grbl/issues](url)
It may explain why you have only got one reply to your question, if you plan to move to the other forum please close this thread and open a new question on the gnea forum.  Sonny hates it when a question has been asked twice, he gets palpitations and all sorts.  He also wants us die hard's to move away from this old forum.