grbl/grbl #1560: abort breaks tools

Arbuthnot999, 2019-10-10:
Got a 3018 CNC, using arduino nano, running grbl 0.9j. Problem is, when I hit abort the machine drives the cutting tool down and then snaps it off. Is there a fix for this?? This reminds me of the reversed jogging (needed $3=5 or similar).
Also bought some limit switches and fitted them. Can someone give me a clue about connection to the arduino??

MeJasonT, 2019-10-10:
Arbuthnot999
I did a quick search and found the following.

[image]

wire something like the image below, you don't have to have a switch at both ends as you can use in tandem with soft limits which will prevent your machine going further than the limits of your bed anyways.  If you can install max limit switches - why wouldn't you.
the image below is for the uno but armed with the pin numbers above you should be able to see how to wire the switches to the nano.

[image]

closed 2019-10-11 by Arbuthnot999

Arbuthnot999, 2019-10-11:
Thanks for all your good efforts, Jason.
I do have an Arduino but it is being used for something else.
Proper name of this board is “Developower v2.0 (pic attached) I hope this shows the info you need.
As you might be able to tell, I’m brand new to forums and discussions. I blame this for my lack of etiquette .
The problem of breaking tools hasn’t been addressed yet. Will this be a code problem or something else??
Once again,
Thanks for your time,
Regards,
Pete Smith (Arbuthnot999)

From: Jason Thompson
Sent: Thursday, 10 October 2019 7:55 PM
To: grbl/grbl
Cc: Arbuthnot999; Author
Subject: Re: [grbl/grbl] abort breaks tools (#1560)

Arbuthnot999
I did a quick search and found the following.

wire something like the image below, you don't have to have a switch at both ends as you can use in tandem with soft limits which will prevent your machine going further than the limits of your bed anyways. If you can install max limit switches - why wouldn't you.
the image below is for the uno but armed with the pin numbers above you should be able to see how to wire the switches to the nano.