grbl/grbl #1561: What happens if the sum of maximum speeds needs more than 115200 Hz?

MARIOBASZ, 2019-10-14:
I have this doubt: what happens if the sum of maximum speeds needs more than 115200 Hz?
suppose that for G0 it gives me 600RPM and I use 1/32 micropasses, it is equivalent to 64000 pps or 64khz.
If I want to simultaneously move two axes at maximum speed, what is the problem?
Synchronization problem / positioning error?
or
Will they run slowly but there will be no problems of trampling?
Should I modify baudrate?
Is my reasoning correct?

langwadt, 2019-10-14:
the maximum step rate is determined by how fast the code can run, not the baudrate
maximum for grbl is roughly 30kHz

langwadt, 2019-10-15:
it is not an exact number it depends, but notes in the source code and other places say the highest safe speed is roughly 30kHz, so your 'steps/mm' * 'mm/min' * 60 should be less that 30k for all axis

zohargolan, 2021-01-15:
I am confused, if I have 80.645 step/mm, according to this formula, I can only have a rate of 6mm/min?????
What am I missing?

AlexHolden, 2021-01-15:
I'm not sure how you came to that. 30,000 steps per second / 80.645 steps per mm = 372 mm per second = 22,320 mm per minute.

zohargolan, 2021-01-15:
Well this sounds much better :-). The formula above is wrong"...  'steps/mm' * 'mm/min' * 60 should be less that 30k ..." should be 'steps/mm' * 'mm/min' should be less that 30k ..." 
Thank you for clarifying @AlexHolden