grbl/grbl #1563: GRBL: inaccurate on X axis by ~4mm on 40cm

element2, 2019-10-21:
This is GRBL-related. I have a laser-engraver type machine, Chinese, ~55 cm X axis, aluminium extrusions, 2mm-pitch T belt, Nemo 17H3430 motors, one mechanical homing switch for X, Arduino-based.The X-axis accuracy is not up to par. When I make a mark on the X axis and jog to that mark (after homing), depending on the day, the position varies within ~4mm or so from G00 X-418 to G00 X-422. I checked the belt tension, it is seems OK, not too tight, not too lose. Is there something else I should check? Another idea I have: put a homing switch on the other end of the X-axis, then home both ends of the X axis, measuring the difference in machine position (assuming I can measure it), then modify the steps/mm value for each session to force to command that puts me on the mark to be constant (e.g. G00 X-420). Any other ideas? Thanks.

jahnj0584, 2019-10-21:
Put sharpie on your pulleys to see if they're slipping. Reduce accel and
top speed and try again.

It's not a software issue.

chamnit, 2019-10-21:
Depends on what kind of error you are getting. If it’s a fixed error regardless of you how much you move, it’s likely a loose set screw on your stepper or lost steps from too high acceleration/max speed. If the error scales linearly with distance, you haven’t programmed the step/mm setting correct.y