grbl/grbl #1564: Can't upload grbl v0,9 or over version in Arduino Nano board

kch1027, 2019-10-22:
Hello

I'm frank, has big problem for uploading grbl v0.9 or over version.
I was used my CNC machine as v0.8 version, hence has tried upload v1.1
However, Arduino nano board can't upload v0.9 or over version just stop or freezing.
I bought other nano board from other seller 3 times but all of them has same problem. 
Therefore changed USB cord, re-burning  bootloader ETC...has .tried everything after searching web.
Only show below error massage. It always fail just before ending the upload with this protocol error.
Please help me someone who know what's wrong.

### I bought Arduino UNO board, It's fantastic upload and also can load hex file through xload
But Nano board only happened this error upload..

[image]

kch1027, 2019-10-22:
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\kch10\AppData\Local\Temp\arduino_build_308822/grblUpload.ino.hex"
avrdude: writing flash (29762 bytes):

Writing | ################################################
avrdude: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x65
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync

MeJasonT, 2019-10-23:
first thing you will be asked is to see if you can load another sketch to your nano to prove its working and the comms/bootloader are working properly.

Are you loading from a sketch using the ino or are you uploading the code as a hex file

I suspect the issue is with the device you have selected in the IDE, if you are using an old nano you cant use the new Nano hardware/device settings

I tried like hell to attach a 40k jpeg but GITs import control is trying to destroy my mood.
So in the arduino IDE
under tools
select your board Nano
and directly beneath you will see a selection for processor
there will be 3 listed 
try ATMEGA328P (old bootloader)

see if that helps
its a gotcha that has caught me out before

kch1027, 2019-10-24:
Hello 
Thanks Mejasoin  for your answer.

During the 4 weeks, I did everything. changed USB cable, re installed driver, re install IDE, tried to use other computer, changed setting under tools of IDE....................................etc. all of 3 Nano board has same error and could't upload grbl v0.9 and above version.
But other sketch were no problem upload, just like "Blink", and tried EEPROM clear, read, update everything fine, good lard, only problem does not upload v9.0 or above version(v1.1)

For no more getting stress and for keeping good my mental health

I thrown Arduino nano board to rubbish bin, and now using Arduino Uno board with CNC shield board. It's perfectly upload v1.1 and fantastic working well
I suspect this problem
1. Grbl v0.9 and above version does not support to Arduino nano v3.0 board
2. Chinese clone microchip has problem

Thank you and apologized  my short English

IonicBob, 2019-10-28:
kch1024, I also have what appears to be the same issue. 
I have a nano (V3, CH340G, optiboot, 115200 baud) and it also times out at the end of write when I try and upload GRBL 1.1e  (29988 bytes) using USB methods Arduino IDE 1.8.10, XLoader or T2laser. I have no trouble uploading (and running) smaller firmware such as GRBL 0.9 (27666 bytes, or any small examples) by any USB method. 
After some experimentation I am speculating this is an issue with this particular flavor of nano (I have several the same), as I can upload the same GRBL 1.1e onto other 328p such UNO or Duemilanove.  I have ordered a programmer so I can try other loaders, speeds, BOOTSZ, etc.  The GRBL 1.1e image I have is small enough to fit into flash with the original nano bootloader (2k). I have also ordered some other flavors of nano's to try.

Arduino: 1.8.10 (Windows 10), Board: "Arduino Nano, ATmega328P"
Sketch uses 29988 bytes (97%) of program storage space. Maximum is 30720 bytes.
avrdude: Version 6.3-20190619
...
Reading | ################################################## | 100% 0.00s
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\***\AppData\Local\Temp\arduino_build_837129/grblUpload.ino.hex"
avrdude: writing flash (29988 bytes):

Writing | ###############################################avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding

IonicBob, 2019-10-30:
Update 1: Received an (ebay) $3 USBasp programmer. The nano in question has an optiboot and was set for 512bytes as expected.  Someone suggested replacing the bootloader, so using the USBasp I did the Arduino IDE "Burn Bootloader" which also uses optiboot (but not necessarily the same version as originally on the board).  Then attempted to program this nano using standard Arduino IDE usb loader, first Blink and then GRBL 1.1e. Blink worked and it got a bit farther through on GRBL, but failed on verify (this subtle change in behavior may be a clue).  Testing it anyways, on the com port the nano outputs the GRBL 1.1e banner, but then crashes.  Next I went back to the USBasp programmer using avrdude and successfully uploaded and verified using the same hex file (another clue).  Then I tried the nano on LaserGRBL, fixed the configuration and then successfully ran a job.  Using the $3 USBasp programmer is not a hassle and it seems I can use these boards. But still leaves me wondering what the problem is with this nano not able to load large firmware on USB (hardware or compatibility issue).

There are several avenues here. There has been chatter in the forums about errors at high baud rates, I don't know if its relevant but its easy for me to test. I could bypass the CH340G or I could try a bootloader at 56700 baud, but can't be bothered making a custom version of optiboot.  I tried the original bootloader which operates at 56700 and although it's larger at 2k, my GRBL 1.1e should still just fit.  Alas the lower speed makes no difference, still same verify error.  I might try the latest version of optiboot, but somehow I don't believe it will make any difference unless they have identified a compatibility issue.  I can't help but wonder if this nano using an Atmel knockoff and is not quite the same. My other flavors of nanos should be here in another day or two, meanwhile I will continue testing my GRBL 1.1e burned with the USBasp.

MeJasonT, 2019-10-30:
I had the problem a while back when i was thinking of using a Nano and as far as i can remember the information i suggested above worked for me.  At the tie i looked into the issue the Arduino IDE had just been updated and there was a lot of chatter on the web regarding the change to the bootloader speed - that is why the old bootloader is in the processor list in the IDE.  i dont know why its not working for you guys.  Also there are several reports regarding the ftdi clone chips such as the CH340G.  Its only five/six years ago that FTDI tried to block the market of cloned ICs by teaming up with Microsoft and disabling drivers which received the wrong ID's from USB chips.  You may have the remains of this issue lurking around inside your PC.  Although if it loads basic sketches then this is highly unlikely.  if i get a chance i will try loading a nano again and see if i get problems i can try to solve and report back.

IonicBob, 2019-11-02:
I'll summarize where I'm at now:
- My clone nano V3 (which has a CH340 USB) mostly operates just fine.  
- There have been no issues with the driver or USB bootloader for any Arduino examples or 'small' GRBL sketches.  For me this includes a GRBL 0.9g sketch. Realize that this sketch can be compiled with different options which change the size. The size of my 0.9g is 27666 bytes. 
- However I have a particular flavor of nano V3 which won't load larger sketches such as GRBL 1.1e which is  29988 bytes, and kch1027 has a build of 0.9 which is 29762 bytes which also won't load via the bootloader.
- I have determined that for this flavor of nano V3 that the threshold for the bootloader not working is 28672 bytes, ie., it bootloads fine except for the last 4k bytes. I thought this might be a problem with erase, but even doing a chip erase or using other bootloaders ends up with the same result.
- But I can load large sketches into this particular nano v3 using the USBasp programmer without any problems and it runs just fine. I have been running this nano using GRBL 1.1e for several days doing multiple passes of very fine detail and it hasn't missed a beat. So its only something to do with the bootloader on this chip.
- I have loaded larger sketches (ie., GRBL 1.1 at 29988 bytes) into other boards, including a different flavor of nano V3 (using a CH340) using the USB bootloader it came with (orig), as well as UNOs and Duemilanove without any problems. Its supposed to work.
- So I am concluding this problem is restricted to just particular flavors of nano V3 and has nothing to do with the CH340.  
- While this m328P has the Atmel markings I have no way to accertain that it is authentic. Maybe this version needs different fuse settings. Or maybe the bootloader just doesn't work in the last 4k even with optiboot and a 512byte bootsz.  I'm not too broken up, its not like I just bought 10000 for a production run. I'm happy to use the USBasp programmer or another flavor of V3. Just a bit of time wasting and curiosity.

MeJasonT, 2019-11-02:
Just as a point of reference
there are no genuine Nano's which use the CH340, to the best of my knowledge all the genuine arduino's use ftdi USB ICs.
The chances of buying a dodgy, dodgy Asian import are going to be quite high.
2x dodgy does not make a good one, the most pants statement in math, two wrongs don't make a right nor does - 25 added to -25 make +50.
I have not come across an Atmel Atmega clone yet that has not worked, so i would argue that our friends in Asia have managed to clone a near perfect copy but its like light bulbs they only batch test these products so there is always a high possibility of getting a duff one in a batch.
you still didnt confirm if you had used the old bootloader option from the IDE.