grbl/grbl #1565: setting up the spindle enable circuit  with an arduino relay for plasma cutting.

McKlay, 2019-10-23:
i really hope someone in here can help me with the complete setup between the ugs controller, cnc v3 shield and the arduino relay.

i've read a lot of forums already, some says i can connect the input pin on the relay on the "+z  limit switch input" since starting at grbl v.9 there was a modification on pins 11 & 12. i set up that circuitry using the normally open (NC) on the relay connected to an led bulb just for testing. but seems like relay is already turning on even though i've just plugged in the power cable for arduino. Some instructed to setup something in the grbl software og in ugs.
i'm searching for clear instruction but seems likw i'm just getting confused. 
i hope someone can help me and give clear instructions on how to set up this.

regards

MeJasonT, 2019-10-23:
The method i have used is to set the spindle out put to max 1000 rpm (with setting for max rpm set to 1000 also) The plasma trigger relay is operated when the spindle output is 1000 rpm and off when its zero.
I use cambam as a CAM/postprocessor application and add a startup and end procedure in the post processor to control the plasma torch.

at start
G4 P1     pause for all movement to end
M8         in my case i'm also using the coolant pin to control a pilot arc relay
              - this effectively earths the tip for non frequency start plasma's
(Pilot Ark)
G4 P1     pause to allow pilot arc to engage
M3 S1000    Trigger relay enabled - Arc started
(Trigger)
G4 P0.25  Delay to allow arc to punch through material
M9       Turn of pilot arc- if left on you will race through consumables, once ark is active there is no 
             need for pilot

and M5 at end to turn off arc trigger/relay

it works like a dream and there is no need to mess with GRBL or your controller

MeJasonT, 2019-10-23:
You will get high frequency noise issues using a plasma with your grbl set up, it causes all sorts of firmware freezes, comm issues and false end stop triggering.  There are plenty of reports on this forum to advise you on how to overcome such issues.  Mainly earth all your machine metalwork, frame, gantry etc to a central earth point and if possible earth using a copper ground rod.  not advisable to use you mains supply earth as it can have stray transient voltages floating about which will add to system noise not eliminate it.  setting correct arc height and using good consumables also eliminates some of the noise issues.  when your tips are burnt out or you have not cleaned the metal material you are cutting it causes the noise issues also.  And as i occasionally find out so does not switching on the air from the compressor, you feel such a tit when you forget and the plasma wont strike.  It creates a 12,000 volt energy spark and does nothing - that's enough to upset your controls with HF noise.

McKlay, 2019-10-24:
> The method i have used is to set the spindle out put to max 1000 rpm (with setting for max rpm set to 1000 also) The plasma trigger relay is operated when the spindle output is 1000 rpm and off when [...]

thank you MeJasonT but i'm using grbl post processor in fusion 360 and my main problem right now
is with the circuitry on cnc v3 shield and relay.

MeJasonT, 2019-10-24:
Please close this thread you have opened a duplicate on the same forum.
I have replied on your other post.  I think you tried to take my last suggestion of moving to Gnea but ended up back here.