grbl/grbl #1566: Stepper in cnc machine are vibrating but not moving

agjupiter, 2019-10-23:
Hi ....I have got a problem with my homemade cnc pen plotter...I have made my cnc machine using the arduino uno, cnc shieldv3.0, nema 17 with dev8825 driver grbl v0.8c/ v0.9i/ 0.9j with the respective baud rate I.e, 9600 and 115200 used ide to flash the grbl and grbl controller for sending g code...

The problem is all things are fine in software (I think). When I connect arduino to pc it is fine, I upload grbl ..I opend grbl controller and send g code but the wrong is I mives in visualizer (shows ) but the motor is not moving and making unpleasant sound...

I have adjusted the vref to 0.8 amps(v) power with 12v 1 amp dc...
I think I have wired corrctly...I also change the wire also a1b1a2b2 to a1a2b1b2 thinking it will work but its same as previously...
Plz help to solve the problem....
Plz plz plz

JesConsuegra, 2019-10-23:
It uses to be the motor cabling. Put two leds between a1-a2 and b1-b2 and turn the motor axle by hand. Both leds should light.If not,  swap the cables until both light.

RenatoMourao, 2019-10-23:
Hi, try to decrease the number of steps per minute of your motors, if it is
too high the motor only will vibrate.
Good lucky!