grbl/grbl #1567: Homing no switches??

SketchThis, 2019-10-23:
Like the title says, is it possible to home a GRBL machine with no limit switches?

Let me explain: I have a CNC I'm converting and it homes itself by jogging each axis farther than it can physically go. The machine will hit the end stops and the motors will stall out. Once that happens a "pull off" movement is done, and the axes location is then set.

Is it possible to set something like this up in GRBLs native homing routine? I'm sure I go do it manually with native G Code commands, but I'd like to have the homing buttons actually work...

jahnj0584, 2019-10-23:
There's no motor feedback so no.

Manually jog the machine against the end stop and reconnect.

SketchThis, 2019-10-23:
I'm not looking for motor feedback, this is how it works on this machine with it's current (non GRBL) control:

To home each axis there's a hard stop at Z zero, X 12 and Y 12.

The machine's max Z travel is 3 inches, so on homing it jogs up 4 inches and hits the end stop. It backs off .125" and is zero'd there.

Same for X and Y, they're jogged to 13 inches (1" more than the max travel) so they hit the end stop. After that 13" move they're jogged negatively 12" and zero'd there.

Does that make sense? These are open loop steppers so no motor feedback is needed. What I need is a homing routing that can function without limit switches. In reality this is just a simple macro... I just don't know if homing on GRBL can be customized like that?

SketchThis, 2019-10-23:
Sorry I realized I posted this in the older version of GRBL, I've reposted in the newer version:

https://github.com/gnea/grbl/issues/746