grbl/grbl #1568: setting up the spindle enable circuit with an arduino relay for plasma cutting

McKlay, 2019-10-24:
i really hope someone in here can help me with the complete setup between the ugs controller, cnc v3 shield and the arduino relay.

i've read a lot of forums already, some says i can connect the input pin on the relay on the "+z limit switch input" since starting at grbl v.9 there was a modification on pins 11 & 12. i set up that circuitry using the normally open (NC) on the relay connected to an led bulb just for testing. but seems like relay is already turning on even though i've just plugged in the power cable for arduino. Some instructed to setup something in the grbl software or in ugs.
i'm searching for clear instruction but seems like i'm just getting confused.
i hope someone can help me and give clear instructions on how to set up this.

regards

MeJasonT, 2019-10-24:
captured from your other post

> thank you MeJasonT but i'm using grbl post processor in fusion 360 and my main problem right now
is with the circuitry on cnc v3 shield and relay. i'm trying to connect the input pin of the relay on the +z-axis input pin on the shield but it seems like the relay is always on and when i tried to run a gcode file in ugs, the led i'm testing doesn't work which means i cannot determine if it's trigger on/off because it's always on. <

Using Z Axis will cause you issues it will only trigger when Z moves are being transmitted, ie 10 mm = 750 steps for for example your relay will pulse 750 times then stop.  You would need to re write grlb to have that pin do something else.  its easier to use the spindle and coolant pins.
the pwm output on the spindle pin confuses some people but if you send max speed = max pwm its pretty darn close to 5v output, if you send 0 then its 0.

MeJasonT, 2019-10-24:
be careful which version of grbl firmware you upload if this board was manufactured when grbl was pre v0.9, if you upload the grbl firmware v0.9 onward the the pin configuration changes for spindle speed, direction and Z end stop.  if could be you are looking for the output on the wrong pin.
the pin changes were made in v0.9 to accommodate a variable spindle speed.

McKlay, 2019-10-24:
> be careful which version of grbl firmware you upload if this board was manufactured when grbl was pre v0.9, if you upload the grbl firmware v0.9 onward the the pin configuration changes for spindle [...]

that you so much, i think i know about that, i'm uploading grbl v1.1 so and i believe it's pin configuration with respect to pins 11 and 12 is same as v.9 as stated from this link https://github.com/grbl/grbl/wiki/Connecting-Grbl

McKlay, 2019-10-24:
> captured from your other post
> 
> > thank you MeJasonT but i'm using grbl post processor in fusion 360 and my main problem right now
> > is with the circuitry on cnc v3 shield and relay. i'm trying to connect the input pin of the relay on the +z-axis input pin on the shield but it seems like the relay is always on and when i tried to run a gcode file in ugs, the led i'm testing doesn't work which means i cannot determine if it's trigger on/off because it's always on. <
> 
> Using Z Axis will cause you issues it will only trigger when Z moves are being transmitted, ie 10 mm = 750 steps for for example your relay will pulse 750 times then stop. You would need to re write grlb to have that pin do something else. its easier to use the spindle and coolant pins.
> the pwm output on the spindle pin confuses some people but if you send max speed = max pwm its pretty darn close to 5v output, if you send 0 then its 0.
> 
> It would leave your z axis free to use for torch height adjustment- for thicker metal etc

thank you so much for quick reply MeJasonT
so with grbl v.9 or v1.1 in which i use the v1.1,  pins 11 and 12 are being swapped to access the hardware PWM on D11 for variable spindle PWM output.  But i'm using the CNC shield V3, that means that i cannot access the spindle enable on the shield rght? and so saying i can use the spindle or coolant pins, do you mean the spindle direction or enable?

MeJasonT, 2019-10-24:
this shows the pin changes as they relate to an arduino uno
[https://github.com/grbl/grbl/wiki/Connecting-Grbl](url)
i'm trying to find a schematic of the CNC sheild V3 i will take a look and hopefully point you to the correct pins

MeJasonT, 2019-10-24:
as far as i can remember the coolant pin remained un changed, sending either M7 or M8 should switch it on and M9 off.  either M7 or M8 wasnt available in early releases of Grbl v1.1 but was added later
they are flood and mist coolant, mist coolant was added to the release no so long ago.

MeJasonT, 2019-10-24:
[https://forum.protoneer.co.nz/viewtopic.php?t=817](url)
seems to cover it
ignore the mistake they have made M5 stops the spindle not M4

MeJasonT, 2019-10-24:
ok i'm with you now using Z+ end stop pin (you didn't mention end stop so i assumed wrongly)
your end stop now becomes pin 12 your spindle En.
The coolant output pin A3 is unchanged so it should work no matter what.
[image]
[image]
[image]

McKlay, 2019-10-24:
> ok i'm with you now using Z+ end stop pin (you didn't mention end stop so i assumed wrongly)
> your end stop now becomes pin 12 your spindle En.
> The coolant output pin A3 is unchanged so it should work no matter what.
> [image]
> [image]
> [image]

i'm using the z+ end stop but yet it's turning on already so i think there's something i have to edit in the UGS first to switch it off. I'm tried tweaking the gcodes through notepad and i wonder why "S1000 M3" which i think should turn it 'on' but instead it turns off at "M3" and turns on at "M5 S1000". Which can be shown from the gcodes below. There's should be something i need to invert or what.

//input pin on the relay is connected to +z end stop (the left side betweeen two pins)
(2D Profile1)  //tested led is on by default
G54
M5 S1000 
G0 X-76.509 Y-24.358
G1 Y-14.358 F1000
G3 X-77.509 Y-13.358 Z0 I-1 J0
G3 X-95.424 Y-31.274 I0 J-17.915
G3 X-59.593 Y-31.274 I17.915 J0
G3 X-77.509 Y-13.358 I-17.915 J0
G3 X-78.509 Y-14.358 I0 J-1
G1 Y-24.358

M3                            //the led tested turns off

G0 Y11

M5 S1000                // the led tested turns on again

G1 Y1 F1000
G3 X-77.509 Y0 I1 J0
G1 X-21
G2 X0 Y-21 I0 J-21
G1 Y-41
G2 X-21 Y-62 I-21 J0
G1 X-85.238
G2 X-85.49 Y-61.968 I0 J1
G2 X-82.865 Y-0.014 I7.982 J30.694
G1 X-82.781 Y-0.004
G1 X-82.696 Y0
G1 X-77.509
G3 X-76.509 Y1 I0 J1

M3                             //turns the tested led off

G1 Y11
G1 S0
M30
%

anyways now i know the +z endstop pin works fine.
also i wanna ask, referring to the picture of the cnc shield above, are the yellow and blue pins on the left side where the endstops, coolant, spindle pins are located the same? cause when i tried to connect the relay input on the blue(right ) side, it doesn't work.

thank you so much for keeping up with me.

MeJasonT, 2019-10-24:
Pin 12 will be using the internal pullup resistor in the arduino keeping the pin high try inverting the Z end stop  pin setting in the grbl configuration settings.

McKlay, 2019-10-24:
> Are the yellow pins the signal connection and blue its associated ground pin
> 
> You can change the behaviour of the spindle in the config.h
> 
> `// Inverts the spindle enable pin from low-disabled/high-enabled to low-enabled/high-disabled. Useful // for some pre-built electronic boards. // NOTE: If VARIABLE_SPINDLE is enabled(default), this option has no effect as the PWM output and // spindle enable are combined to one pin. If you need both this option and spindle speed PWM, // uncomment the config option USE_SPINDLE_DIR_AS_ENABLE_PIN below. // #define INVERT_SPINDLE_ENABLE_PIN // Default disabled. Uncomment to enable.`
> This might put the pins back to the old version firmware allowing you to use the pins on your board as they are labled - you dont need variable spindle for the plasma trigger relay.
> Are you confident changing the code?
> 
> A quick tip for testing dont power up the cnc board with external power
> your pc will be more than happy talking the the arduino all day long pretending the machine is connected, the only frig is to disable hardware limits and homing in the grbl settings - obviously switch them back on when you connect the machine or you wont be able to home or warn of over travel

1.) Does editing the config.h and disabling the PWM means instead of "M3 S1000" i'm just gonna use "M3" to trigger it on and off?
- anyways other choice is that i can just connect my plasma torch on the normally close pin instead of normally so i can use M3 S1000 for turning it on and M5 for turning it off and not the other way around.

- also since it was swapped, i tried my z-endstop on spindle enable pin connected to a limit switch for z-axis and it does work.

2.) Next thing i'm working on is the homing, i want to set home when 3 limit switches are triggered but then to be offset at the center of my wokspace where it should be the (X0,Y0,Z0) reference point, do you have any tips for that like the first home can be done automatically maybe by setting in UGS, or do i have to edit that gcode manually? 

 thank you so much for your help, God bless you.

MeJasonT, 2019-10-24:
Set homing to 1 in grbl configuration settings
grbl has a built in command to home all axis
$H run homing cycle
Theer is some information on setting it up here
https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle

i think you have to send an M3 and S*** command anyways M3 will trigger the direction pin and S*** will enable.  It will be interesting to find out how this behaves as i have no knowledge of it.  my machines (yes more than one) have always worked as wired without me having to look too deeply into it.

The pin inversion has been mentioned on the forum before (pin 12) but i have not mastered how to search on GIT yet, I never get the results i'm looking for ie terms in old threads
perhaps cpu,.h which lists the pin configuration will have a clue in there somewhere.
I spent some time this afternoon looking for you but no luck, i'm sure there used to be a pins.h but i may have imagined it.

McKlay, 2019-10-28:
> Set homing to 1 in grbl configuration settings
> grbl has a built in command to home all axis
> $H run homing cycle
> Theer is some information on setting it up here
> https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle
> 
> i think you have to send an M3 and S*** command anyways M3 will trigger the direction pin and S*** will enable. It will be interesting to find out how this behaves as i have no knowledge of it. my machines (yes more than one) have always worked as wired without me having to look too deeply into it.
> 
> The pin inversion has been mentioned on the forum before (pin 12) but i have not mastered how to search on GIT yet, I never get the results i'm looking for ie terms in old threads
> perhaps cpu,.h which lists the pin configuration will have a clue in there somewhere.
> I spent some time this afternoon looking for you but no luck, i'm sure there used to be a pins.h but i may have imagined it.
> more info here
> https://bitraf.no/w/images/4/4c/GRBL_software_Instructions_CNC3-3018Pro.pdf

i'm so happy it worked, thank you!
last thing i wanna ask is the circuit diagram for the noise filtering connected with the shield and normally close of the switch.
i'm thinking about this ...
https://cloud.githubusercontent.com/assets/5912573/22625452/1671414a-eba0-11e6-9fb1-648a82bd19bf.png 
but when translating in the shield there could be some errors so maybe you have something useful like a circuitry which includes a the V shield with some resistor and capacitor values.

MeJasonT, 2019-10-28:
don't bother with the three pull up resistors as the board is more than likely using the internal pullups
the 100 nf caps in the picture are the correct values - just wire as shown.
the capacitor value will have a lag based on its value at 100nf is unlikely to be much different to the action time of your switch.  Its a tried and tested method and gives little or no error in machine operation.  if you do have a lot of grbl firmware crashes and the feedback messages from grbl are indicating its the end stops.  disable the hardware end stops in the grbl settings.  despite this the machine will still home using the hardware switches but will ignore them during normal operation - don't use as a permanent fix though, try and find any problem and get rid or it.  use screened cables for both your motors and end stops. Earth all metal surfaces back to a common earth point (individually - don't be tempted to daisy chain them).  you should be up and running in no time.
there is one more fix for excessive em noise but it should only be carried out if you are extremely competent.  set the arc gap inside the plasma unit itself, many cheap Asian imported machines have the gap set with anything ranging from too small producing lots of noise and an under powered cut to too big producing nothing but noise - This is a last resort and not for the faint hearted.

is your plasma a high frequency or strike start ie does it have a pilot ark ?

McKlay, 2019-10-28:
> don't bother with the three pull up resistors as the board is more than likely using the internal pullups
> the 100 nf caps in the picture are the correct values - just wire as shown.
> the capacitor value will have a lag based on its value at 100nf is unlikely to be much different to the action time of your switch. Its a tried and tested method and gives little or no error in machine operation. if you do have a lot of grbl firmware crashes and the feedback messages from grbl are indicating its the end stops. disable the hardware end stops in the grbl settings. despite this the machine will still home using the hardware switches but will ignore them during normal operation - don't use as a permanent fix though, try and find any problem and get rid or it. use screened cables for both your motors and end stops. Earth all metal surfaces back to a common earth point (individually - don't be tempted to daisy chain them). you should be up and running in no time.
> there is one more fix for excessive em noise but it should only be carried out if you are extremely competent. set the arc gap inside the plasma unit itself, many cheap Asian imported machines have the gap set with anything ranging from too small producing lots of noise and an under powered cut to too big producing nothing but noise - This is a last resort and not for the faint hearted.
> 
> is your plasma a high frequency or strike start ie does it have a pilot ark ?
> 
> Just for info i did discover that the coolant enable pin can be inverted in config.h to be active low instead of active high.

Yes i used to have crashes when enabling the hard limits but i did find ways in ugs on how to fix this.

But i did include the resistors, i guess its will still works.

Yes mine is a high frequency start so i guess i'll be having more EMI problems, but i setup already a DYI faraday cage and i guess shielding the motors, and limit switches wires with aluminum foil could be enough to block the EMI, i'm from Philippines and in here i really cannot find some shielded heat shrink so i'm just using aluminum foil.

Thank you so much Jason, your help for our thesis is so much appreciated.
God bless you.

MeJasonT, 2019-10-28:
This is the tackle you need
https://www.ebay.co.uk/itm/Stainless-Steel-304L-Over-Braid-Sleeving-Rubber-Hose-Cable-Protection-Etc/123903159109?var=424662391187&hash=item1cd9342b45:m:mpjR1R-J_L-GTJr0-2H36Mg

MeJasonT, 2019-10-28:
only connect braid at one end to earth point, 
Don't do anything at other end or you will create earth loops (a huge radio antenna).

McKlay, 2019-10-29:
> only connect braid at one end to earth point,
> Don't do anything at other end or you will create earth loops (a huge radio antenna).
> 
> which province do you live in Manila, Cebu ?
> 
> Here is a plasma table I made
> https://www.youtube.com/playlist?list=PLutvz5Ge_F7Ntf28XIZAeAq5G0c21bKEb

I live in Cebu.

McKlay, 2019-10-29:
> This is the tackle you need
> https://www.ebay.co.uk/itm/Stainless-Steel-304L-Over-Braid-Sleeving-Rubber-Hose-Cable-Protection-Etc/123903159109?var=424662391187&hash=item1cd9342b45:m:mpjR1R-J_L-GTJr0-2H36Mg

so you mean it is better than wrapping the wires with aluminum foil?
or i should do both like placed the wrapped wire(of aluminum foil) inside the stainless steel braided sleeves?

MeJasonT, 2019-10-29:
No foil needed, just he braid.  if its not full and fully expanded then it closes up to  almost a space craft type armouring.  It helps to reduce wear if you are dragging and flexing your cable harness and provides some mechanical protection as well.
just a hint it wont stop the plasma cutting through it or protect the cable harness from much heat - that's your job.
I worked on a pipe laying construction vessel in India for a company called Allseas, many of the crew cam from Cebu, I made many good friends and literally got a tee shirt from Cebu.  The guy running the laundry left to spend time with his wife and new born baby, we had a whip round on board for him and gave him some money to help his young family.  The tee shirt was to say thank you.
I don't worry i don't expect a tee shirt for helping you, i am just sharing a story of the wonderful and friendly people i have met.  Its as if you were there you are now held in high regard as a friend like those guys.  I look forward to seeing your project succeed.

MeJasonT, 2019-10-29:
On a side note
It looks like you can send command M3 S1100 or S1100 M3 and grbl forgives you.
S1100 M3 makes logical human sense set the speed then activate the spindle
but from a computer programmers point of view, in coding the answer always returns to the left
if you move data into a variable you would put ansX = a+b and not a+b=ansX obviously there are more complex examples but that's where i'm coming from.
so if you put M3 S1100 you are telling the machine to run at speed and then activate (returning to left).  There are tons of gcode programming tutorials out there and the bias is in favour of M3 S1100.
why your inventor fusion 360 post-processor does it backwards i have no idea, especially as the tutorial i started with was written by autodesk ? fools the lot of them.

jahnj0584, 2019-10-29:
Because the cnc interprets the whole line at once,not left to right.
Doesn't matter the order- Same thing for drilling cycles.

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?

guestisp, 2022-04-17:
hi. simple question: did you notice that Y endstop is always powered by 3v3 ?