grbl/grbl #1572: Anyone familiar with the GRBL C code itself? (sync Z servo to XY steppers)

element2, 2019-11-15:
Is there someone who knows the GRBL C code itself? I have an 2-axis machine, GRBL v1.1, with a servo on the Z axis. Up until now, I've controlled the servo with its own, separate line of G-code (using an adopted M3 command). Now I want to sync the Z servo move with an actual XY stepper move. To do this, I've changed the GRBL C code in "stepper.c" and "planner.c" that deals with the planner "blocks", the chunks of code that the planner processes. However, I can't quite get it to work, because I don't have a good understanding of what the data structures in those C files do.

cri-s, 2019-11-15:
Yes, but in that case use servo the same as stepper, you have 0.5us
resolution on servo, this means on 2ms (180 degr) you have 4096 steps for
the servo. For 90 degree. This is 2024 steps (timing units).

element2, 2019-11-15:
Using a servo and steppers at the same time isn't a problem. The steppers and the servo are on separate timers (it's an Arduino Mega). I can physically move the servo and steppers at the same. The problem is accurate synchronization: at any given moment, the servo needs to be exactly where the motion planner decides the Z position should be w/r to X and Y steppers. GRBL splits G-code into "blocks" and "segments" and processes them one-by-one from a planner buffer. So all the instructions for the servo need to be "smuggled" into each of these "blocks". This requires an understanding of the planner and its data structures.

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?