grbl/grbl #1573: Error upload GRBL

GillisBoard, 2019-11-16:
Hello

I'm trying to upload GRBL 1.1 and this displaying a strange error, I've already checked the cable, port COM  and the selection of the Arduíno Uno board on the Arduíno IDE, trying to solve the problem I also tried to send the file ( .h ) of the grbl using the xLoader but does not work too a, tried earlier versions of the grbl and also does not work, to finish the tests I uploaded the example of Arduíno ( Blink ) and it worked all the problem this being with the grbl. Someone could help me I need my CNC machine to work.

>Sketch uses 29594 bytes (91%) storage space for programs. The maximum is 32256 bytes.
Global variables use 1484 bytes (72%) of dynamic memory, leaving 564 bytes for local variables. The maximum is 2048 bytes.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -carduino -PCOM4 -b115200 -D -Uflash:w:C:\Users\GILLIA~1\AppData\Local\Temp\arduino_build_426386/grblUpload.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\GILLIA~1\AppData\Local\Temp\arduino_build_426386/grblUpload.ino.hex"
avrdude: writing flash (29594 bytes):

Writing | ################################################
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding

Problem loading to the card. See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
--

Thank you

BigALDesigner, 2019-12-16:
I have same problem. Machine was working fine and was using Openbuilds but now I get "No supported Firmware"

MeJasonT, 2019-12-16:
Things to ask yourself.
what version of the Arduino IDE are you using, have you recently updated or reinstalled it
you mentioned that you have loaded blink and it worked so coms are ok

I vaguely remember an issue like this years ago but with recent IDE versions its never been seen again.

you sound like you know what your doing so its unlikely you have not selected the port or processor

umm  interesting,  i'm going to scour the web and see what i can find out.

BigALDesigner, 2019-12-16:
New to all of this. CNC Router machine comes standard with Arduino. Uploaded, got it working for 3 weeks only issues for me was that I was using Easel and therefore needed to be connected to the web/cloud so tried loading a few other offline senders as the local power utility and internet are not up all the time. Problem started after I loaded CNCJS.

MeJasonT, 2019-12-16:
@BigALDesigner does Easel not load its own firmware version of GRBL to the controller ?
I remember something weird like that.  if that's the case other GUI's wont be able to communicate with it - its the way they decided to handshake the comms between the device and PC

MeJasonT, 2019-12-16:
@GillisBoard 
It sound like you are using a cloned arduino board, its common for 1 in a 1000 boards to be challenging, they are not exactly built to a fantastic standard and using poor quality faked micros.

However don't give up,
I found this interesting [https://forum.arduino.cc/index.php?topic=415264.0](url)

There are plenty of us quite happily using cloned boards, only once have i had to give up on a board.

BigALDesigner, 2019-12-16:
@MeJasonT I ran Easel and Openbuilds one after the other and there werent any issues. It is only after I loaded CNC JS that the message below started appearing on the Openbuilds controller.

"ERROR!: No supported firmware detected - you need a controller with Grbl 1.1x on it, or there is a problem with your controller. Closing port COM3 [13:17:58] [ disconnect ] PORT INFO: Port closed"

Now I cannot get Easel to run either. I have uninstalled and re-installed everything and still same.
I have attempted the grbl upload but no luck either.

BigALDesigner, 2019-12-16:
I tried using the Openbuilds flashing tool but..... 
[image]