grbl/grbl #1583: Homing return Alarm when the limit switch is engaged

malicmt, 2019-12-10:
Hello,

I'm new to grbl. When I tried to set-up my first CNC which I made using Arduino Uno and CNC shield, I'm facing the following issues.

1. When I enable homing and perform a homing sequence, it is showing Alarm when my Z-axis limit switch is engaged and the homing will be stopped.
[image]

2. When I enable soft limits, I cannot jog in positive direction.
[image]

Attached images will give more details about my settings and the issues.

928oleg, 2020-02-06:
Setting $3=7 and $23=7 solved the problem. Thank you all very much for your help, it is appreciated.

928oleg, 2020-02-06:
To be honest, I also can’t set a positive limit, help me find out how to do this?

roboter, 2020-09-16:
Same here, homing works, but soft limits can't be set, no movement.
I also have switch to X- then Move should allow me to move in positive direction.