grbl/grbl #1584: Hi , I'm new here so if I screw up please bare with me

SeeMyCat, 2019-12-11:


SeeMyCat, 2019-12-11:
I have problem with small CNC machine. It came with an uno controller and works but steppers go crazy
in the middle of a job at any time. I switched to a uno clone which also works but it will not turn on the spindle. The original board is a real Uno and the clone is a FreeNove Uno. So two problems. Have another clone on the way incase it is a bad one I'm trying to use. Any ideas on either problem I'm having?

X3msnake, 2019-12-11:
you must be sure the shield pins are the correct ones..most chinese
machines used shields that where designed for the grbl 0.8c

you probably installed the latest version on your own uno and that is the
reason the spindle is not turning on.

confirm the pinouts first and read the docs on how to configure the pinouts
in case it is the 0.8c issue

SeeMyCat, 2019-12-12:
Hey thanx X . Sounds logical and yeah I did use newest version. Don't know who's shield it is and says nothing on the board such as brand or maker. Shows pin 4 as spindle enable. Not sure how to change them anyway. Shield is Red if that helps. LOL. Guess I'll see if I can find 0.8c and reburn to see if that works. Thanks again.

SeeMyCat, 2019-12-12:
Seems you was right. Works on 0.8c. I can at least see if other problem goes away now. Thank you very much sir.

SeeMyCat, 2019-12-12:
Update:  Other problem seems to be gone now. So it looks like original Uno that came with machine may have an internal short or something.