grbl/grbl #1585: Takim degistirme

hsyngndz85, 2019-12-13:
Slm burda yeniyim grbl otomatik takim degistirme islemi icin cnc js uygulamasinda makro yazilabiliyor.ama bi sorun var 2 adet kesme pini şart bu iş için. 1. Pin frezenin çalisip çalismadiginin kontrolu digeri freze klidinin durumunu izlemek icin kullanilacak. Hali hazirda frezenin calistigini kontrol edilmesi sart eger freze tam calismadan malzemeye dal8nirsa cok tehlikeli kazalar cikar. Eger bu iki guvenlikte eklenirse bu grbl alt yapisi tam profesyonel hal alacak.

hsyngndz85, 2019-12-14:
Slm here I'm new grbl for automatic tool change process can be written macro in the cnc js application. But there is a problem with 2 pcs cutting pin is a must for this job. 1. Check whether the pin cutter is operating to monitor the other milling cutter in place. It is necessary to check that the mill is in operation and if the mill is immersed in the material before it is fully run, dangerous accidents will occur. If these two safety features are added, this infrastructure will be fully professional. The external feed and pause pins can be set as cutting for automatic tool change.

hsyngndz85, 2019-12-14:
isn't there someone who can

MARIOBASZ, 2019-12-21:
Sorry if I do not understand well and what I write is meaningless.
I am advancing in the development of an ATC, I use bCNC, there I generated the macro to change the tools.

hsyngndz85, 2019-12-21:
problem 2 pin queries required

hsyngndz85, 2019-12-21:
Milin durdugunu ve takimin kilitlendiginin sorgusu gerekli

MARIOBASZ, 2019-12-22:
I understood, good idea, redundancy for security, a cut (relay)? or a sensor to control that the spindle is stopped?
I think it is usual to write in English.
Cheers