grbl/grbl #1587: Steppers humming and holding torque, but no movement

herrluki, 2019-12-31:
Hello everybody, 
I am having trouble with my homemade laser cutter...
I am using an arduino nano with a v4 cnc shield. When I move the motors (with LaserGRBL) both of them are humming and holding torque, but not moving. I tried various wiring setups but none of them worked. I switched out the arduino, because I couldn't overwrite it anymore, but that also didn't help.

Please help me,
thanks, Luki

[image]

[image]
[image]
[image]
[image]

jahnj0584, 2019-12-31:
I guarantee you have the motor coil pairs reversed, since you didn't say
you checked that.

herrluki, 2019-12-31:
@jahnj0584 
thanks a lot for your quick answer!
I tried that allready, but I think I didn't try out all variations...

herrluki, 2020-01-01:
I found the mistake, thanks!
the pins were the wrong ones for my arduino nano (in the grbl file)

https://www.davidpilling.com/wiki/index.php/CNCShield

closed 2020-01-01 by herrluki