grbl/grbl #1588: Grbl 1.1 with nano won't work

mindmad19, 2020-01-05:
Hello there
I have upload grbl 1.1 on arduino nano and steeper motor won't move .

But when upload grbl 0.8 it working 
Are grbl 1.1 don't support arduino nano or need to change config ?

Richard1906, 2020-01-10:
maybe

hotzenklotz, 2020-02-18:
I have Grbl 1.1 running on a Nano without issues. Perhaps, this is a config problem.

quotelawrence, 2020-03-09:
i wanted to respond also and say that I received a message for updating the nano grbl 1.1 which cause my software and machine not to recognize or use a laser module which I just purchased I went backwards to version to grbl .9 and is reading the codes, and recognizing and program but I question what is happening and why so many issues with grbl 1.1f, & h,