grbl/grbl #1592: axis direction command problems

Ron1Ron, 2020-01-11:
Universal Gcode Sender (version 2.0 [nightly] / Jan 06, 2020)
Same on all axis problems! 

I enter Y-50 on the command line and hit enter, works fine, I enter Y50 on the command hit enter to return to the same position i started from,  it goes past and hits the limit switch or it travellers in the wrong direction.  with a none specified amount.

I've got this CNC 3 years ago and gave up on it about a year later because of this issue. I've  read and played with everything i can think of but still have the problem. 

Any help would be a  appreciated as I've been trying all over Christmas to see if I can solve it. 

[image]

jahnj0584, 2020-01-11:
DIR pin is loose.

Ron1Ron, 2020-01-11:
I've completely replaced the board and drivers and soldered all the connections in a last ditch attempt to sort it.

Ron1Ron, 2020-01-11:
I forgot to say that the keyboard movement works correctly every time

Ron1Ron, 2020-01-14:
Well not many responses but wasn't expecting any, found the problem sorted it, thanks