grbl/grbl #1593: DUAL_AXIS_CONFIG_PROTONEER_V3_51

demigh0d, 2020-01-12:
Trying to get dual axis working. I'm using the RPi CNC hat from Protoneer which the developer says has the same pinout as the Arduino CNC board.

I have the board set to map the 4th driver to A3/A4 and the endstop is connected to the Z endstop (I'm not using the Z axis pins for anything else).

The Y-Axis motor works but the A-Axis motor never moves.

Here's how my config is setup for it.

> #define HOMING_CYCLE_0 (1<<X_AXIS)
> #define HOMING_CYCLE_1 (1<<Y_AXIS)
> #define ENABLE_DUAL_AXIS
> #define DUAL_AXIS_SELECT  Y_AXIS
> #define DUAL_AXIS_CONFIG_PROTONEER_V3_51

Any suggestions would be greatly appreciated.
Duane

blindMoe, 2020-02-15:
I am in the same boat.  I am using the standard CNC Shield on an Arduino Duo.  One thing I am not certain of is wether or not the A axis should be slaved to the Y axis via the jumpers on the shield.  I have setup dual endstop homing on marlin 2 and in mach3 and never had an issue and would think the drives could not be physically slaved due to them needing to be able to move independently from one another.  Would love to see a wiring diagram / grbl settings write up once someone has this working properly.

demigh0d, 2020-02-15:
Don't slave A to Y. Jumper as shown in the graphic below.. This will connect the A axis to pins on the arduino, (A4/A3 on the rpi shield, might be different on the arduino shield). 

I found that the pins were reversed on the rpi cnc board so  I reversed them in cpu_map.h

[image]

blindMoe, 2020-02-15:
yeah I am not slaving them right now, just can't seem to ever get the A axis to move.  Going to dig up the pinouts for the shield I have and ensure grbl is indeed mapping to the proper pins for the A axis

demigh0d, 2020-02-15:
I finally attached an LED/resistor to the pins and then tried moving it in software. The dir pin will be either on or off depending on the direction it's moving. The stp pin will flicker.

blindMoe, 2020-02-16:
ok if anyone else runs into this and has the yellow and blue CNC shield with 4 driver slots, the proper setup is:
1) No jumpers on the slave setting linking A to any axis
2) I am using GRND and Enabled from the normal drive outputs near the reset switch to drive all the motors
3) X Y Z motors hooked up normally
4) A motor Step/Pulse hooked up to  Spindle Enabled
5) A motor Direction hooked up to Spindle Direction

Config.h changes:
1) Uncomment `#define DUAL_AXIS_SELECT  Y_AXIS`
2) Comment out `#define DUAL_AXIS_CONFIG_PROTONEER_V3_51`
3) Uncomment `#define DUAL_AXIS_CONFIG_CNC_SHIELD_CLONE`