grbl/grbl #1594: Hard Limits Work Correctly, But Homing Fails Error:9

reynolds087, 2020-01-13:
I have the hard limits working now with limit switches, but when I try to do homing, the machine travels to the end of the axis and starts buzzing loudly, but never finds the home position. If I reset to stop the buzzing, it returns an error:9 [MSG:Check Limits].
This is on a woodpecker cnc 3018

All 6 of the switches work as expected when jogging the machine into them, they stop, and I have to reset and unlock the machine to move again. Usually have to do it twice, because it locks a second time when moving off of the switch. I've tested both ends of all three axis with the same results. It's only when I try to home it, then the machine acts like it can't see the switches there and just keeps cranking the stepper motor until I reset.

ondras12345, 2020-01-13:
Error:9 and Alarm:9 [aren't the same](https://github.com/gnea/grbl/issues/111#issuecomment-275931385). Make sure you get it right.

Also, please post your GRBL version and settings.

reynolds087, 2020-01-13:
It's error:9 for sure. Here is the output from the console when I try to home it. Keep in mind, I reset in the middle of it because it goes past the limit switch and just keep cranking the stepper motor.

$H
[CTRL+X] < Grbl 1.1f ['$' for help]
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]
S1000 < error:9

$1=255
$2=0
$3=0
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=100.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=500.000
$111=500.000
$112=600.000
$120=30.000
$121=30.000
$122=30.000
$130=200.000
$131=200.000
$132=200.000

PicEngraver, 2020-01-14:
Are you using a Woodpecker ver. 3.4 board?  If so, the X and Z limit switch inputs are marked backwards on mine.  Swapping the X and Z inputs fixed my homing issue.

reynolds087, 2020-01-14:
> 
> 
> Are you using a Woodpecker ver. 3.4 board? If so, the X and Z limit switch inputs are marked backwards on mine. Swapping the X and Z inputs fixed my homing issue.

Thanks, this fixed my problem. Greatly appreciated!!

PicEngraver, 2020-01-15:
Great.  Thanks for the feedback.

Igor1311, 2021-12-17:
Hi guys, i think my problem is similar to this one.

I am a complete beginer regarding CNC machines. The one i got is 3018 pro. After thinkering a bit with settings etc. Im getting a buzzing noise from motors, but machine is not moving accross any axis, or responding to comands. I am using UGS software.
The buzzing stops once I start homing, but it allways ends with alarms - 
**com.willwinder.universalgcodesender.AbstractController$UnexpectedCommand: An unexpected command was completed by the controller.
	at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:830)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:158)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:851)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:222)
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:244)
	at java.lang.Thread.run(Unknown Source)**

Any chance you could help me with this one?

my settings are

0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 1    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 115    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 1600.000    (X-axis maximum rate, mm/min)
$111 = 1600.000    (Y-axis maximum rate, mm/min)
$112 = 100.000    (Z-axis maximum rate, mm/min)
$120 = 20.000    (X-axis acceleration, mm/sec^2)
$121 = 20.000    (Y-axis acceleration, mm/sec^2)
$122 = 20.000    (Z-axis acceleration, mm/sec^2)
$130 = 300.000    (X-axis maximum travel, millimeters)
$131 = 180.000    (Y-axis maximum travel, millimeters)
$132 = 45.000    (Z-axis maximum travel, millimeters)

Igor1311, 2021-12-17:
This is the Alarm and Error im getting -

[Error] An error was detected while sending '$H': (ALARM:9) Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring. Streaming has been paused.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.