grbl/grbl #1607: PWM spindle control with Electromagnetic Relay

MicroTransactionsMatterToo, 2020-02-14:
So, I've got a cheap chinese CNC, which uses an electromagnetic relay for controlling the spindle. Would this be appropriate to use PWM on. I'm leery of doing so since it's a moving component and I don't feel mechanical relays are typically intended for high frequency toggling.

vigzo, 2020-02-14:
You can't controll a relay by a pwm. :(  The relay can't switch so fast and accurate as teh pwm signal.
You may change the relay to a Mosfet and connect it, by an optocupler, to insilat the circiuits. This onli works if you have a DC motor, not an AC at yor spindle

MicroTransactionsMatterToo, 2020-02-23:
Figured as much. I'll just buy a new control board at some point. The RAMPS one isn't that expensive.

closed 2020-02-23 by MicroTransactionsMatterToo