grbl/grbl #1611: Modify grbl for CNC model

SanZamoyski, 2020-03-02:
Hi and hello,

I know it is NOT an issue, but I need your help as I'm trying to resolve this for two days...
My goal is to change spindle behavior to:
* when it is on - PWM 50%,
* when off - LOW
* when on AND Z-Work Position  < 0, PWM 100%.

I was trying alot of variables and in many places of the code, but with no luck. I've ended with report.c file on line ~522 (under float wco[]) - it is place where WPos/MPos are calculated, before it is printed.

I have this (dirty) code here:
```
    if(current_position[Z_AXIS] - gc_state.coord_system[Z_AXIS]+gc_state.coord_offset[Z_AXIS] <= 0){
        pinMode(13, 1);
        digitalWrite(13, 1);
    }else{
        pinMode(13, 1);
        digitalWrite(13, 0);
    }
```
I've tried 
```
sys_position[Z_AXIS] - (gc_state.coord_system[Z_AXIS] + gc_state.coord_offset[Z_AXIS] + gc_state.tool_length_offset )< 0
gc_state.position[Z_AXIS] - gc_state.coord_offset[Z_AXIS]
gc_state.coord_system[idx]+gc_state.coord_offset[idx]
```
and everything what came to my mind analysing the code, bot none worked.
The effect is that LED (connected to 13'th pin) is on when Machine Position is under zero, not Working Position.

Please, please, please help me. I just need to know is there any variable I can use here, and if not - how to calculate it.

I'm quite sure that sys_position is Machine Position, so [Z_AXIS] should give me what UGS shows under Machine Position. WCO is difference between MPos and WPos, so adding/substracting should give me right value. But it is not somehow...

Best regards...

whitetd, 2020-03-02:
create a post process for what ever your using to create your gcode to do
it, rather then in grbl

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

SanZamoyski, 2020-03-02:
Unfortunately it is not an option. I forget to mention that I'm trying to modify grbl for the CNC model for students. X and Y axes are made with regular steppers. Z and spindle are replaced with laser - that's why I need to use different PWM for over-zero and under-zero. Students will have to program moves by hand (lessons are focused on G-Code, not machine/program). That's why postprocessing option is not an option.

I've already modified GRBL to use h-bridge with microstepping for tho steppers (the third stepper is possible sacrificing some pins but without microstepping) so I would never suspect that this goal was that hard for me to achieve but it is... 

I know that report.c is not the right place to add this functionality. At the beginning I was trying to add this at the end of the stepper movement function but there was already mentioned problem so I moved it here to be sure that this is related to wrongly chosen variables. And it is.

langwadt, 2020-03-02:
could you just add a few lines to the top of the stepper interrupt that sets the pwm to 50/100% depending on whether the last direction was up or down?

SanZamoyski, 2020-03-02:
It would be almost a good solution, but unfortunately, it would set falsely PWM at 100% if "spindle" is over the material. Please keep in mind that it will be used by people who do not know almost anything about G-code, so this kind of scenario are quite possible.
Is this really very complex problem and we need to look for a workaround?

langwadt, 2020-03-02:
I guess something along these lines, 

`float zpos;`
`zpos  = sys.position[Z_AXIS]/settings.steps_per_mm[Z_AXIS];`
`zpos -= gc_state.coord_system[Z_AXIS]+gc_state.coord_offset[Z_AXIS];`
`zpos -= gc_state.tool_length_offset;`

SanZamoyski, 2020-03-02:
I believe I tried something like this by:
`if(sys_position[Z_AXIS] - (gc_state.coord_system[Z_AXIS] + gc_state.coord_offset[Z_AXIS] + gc_state.tool_length_offset ) < 0){ ... }`
based on:
```
  float wco[N_AXIS];
  if (bit_isfalse(settings.status_report_mask,BITFLAG_RT_STATUS_POSITION_TYPE) ||
      (sys.report_wco_counter == 0) ) {
    for (idx=0; idx< N_AXIS; idx++) {
      // Apply work coordinate offsets and tool length offset to current position.
      wco[idx] = gc_state.coord_system[idx]+gc_state.coord_offset[idx];
      if (idx == TOOL_LENGTH_OFFSET_AXIS) { wco[idx] += gc_state.tool_length_offset; }
      if (bit_isfalse(settings.status_report_mask,BITFLAG_RT_STATUS_POSITION_TYPE)) {
        print_position[idx] -= wco[idx];
      }
    }
  }
```
Where print_position is memcopied and coverted by system_convert_array_steps_to_mpos() sys_position. Which is, diggin deeper, `steps[idx]/settings.steps_per_mm[idx]`.
print_position is then used here:
```
static void report_util_axis_values(float *axis_value) {
  uint8_t idx;
  for (idx=0; idx<N_AXIS; idx++) {
    printFloat_CoordValue(axis_value[idx]);
    if (idx < (N_AXIS-1)) { serial_write(','); }
  }
}
```
But I missed calculating to mm. And You are correct. Thank You for pointing that out. Also I thought  that I don't need recalculating to mm/inch or anything. I'll check Your solution tommorow. Thank You.

SanZamoyski, 2020-03-03:
Thanks to @langwadt I found the solution to my problem. In the file report.c I put before `//Report machine position` this code:
  
 ```
 float zpos;
  zpos = sys_position[Z_AXIS]/settings.steps_per_mm[Z_AXIS] - wco[Z_AXIS];
  //zpos -= gc_state.coord_system[Z_AXIS]+gc_state.coord_offset[Z_AXIS];
  //zpos -= gc_state.tool_length_offset;
   
  if(spindle_get_state() != SPINDLE_STATE_DISABLE){
    if(zpos <= 0){
      SPINDLE_OCR_REGISTER = 255;
    }else{
      SPINDLE_OCR_REGISTER = 32;
    }
  }
    
```
Also commented `//SPINDLE_OCR_REGISTER = pwm_value; // Set PWM output level`. in spindle_control.c.

It seems to work very well now without any workaround ;)
Probably there are a better places to put this code but I am not sure if it is needed.

closed 2020-03-03 by SanZamoyski