grbl/grbl #1614: PROBLEM WITH STEPPER MOTOR

davidecnc, 2020-03-09:
good evening.
I have a cnc shield v3.
I put the fw grbl 0.9.
I connected everything according to the scheme.
The motors work correctly because they move and do what they need (I use the sw lasergrbl).
I have a problem, when I turn everything on, the engines are not paired but I can move them by hand.
How can I solve the problem?

abdset, 2020-05-24:
What do you mean by "the engines are not paired" ?
And yes you can move them by hand unless you set $1 = 255, but if you do that motors will be locked but constantly powered even when they don't move and this will result on overheating them.

jahnj0584, 2020-05-24:
Motors don't overheat from holding torque...

Norwegian-Gadgetman, 2020-05-25:
They can overheat from holding torque, but it takes a long time, and you would probably have the power ramped up way too high, anyway.

You often WANT it to be able to hold until you manually disable the steppers. On a machine without homing, it's the only way that you can change tools without losing your position.
And on long straight X/Y cuts, you want the non-moving motor to hold because odds are that it's on a microstep, not a full-step position, and as soon as it loses the holding power it'll slip over to the nearest full-step.
A micro-step here, a micro-step there, and you're soon cutting where you shouldn't.

In fact, if you're cutting a single tools-width, the the friction of the end-mill against the material can be enough to pull the carriage to the side if a motor is disabled.

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?