grbl/grbl #1615: Probing Hard Limit Error

M4RC0660, 2020-03-09:
Hi everyone,
I'm trying to install a probe to my DIY CNC. I connected some wires to the A5 pin (sensing) and to the GND pin of my Arduino UNO. When I bring both wires into contact, thus closing the circuit, I get the desired probe signal as expected (contact with work piece).
My problem starts when I try to attach either one of the cables to the endmill of my router (Makita RT0700). As soon as I connect it the software (bCNC) reports an Alarm:5 "Hardlimits triggered". It doesn’t matter whether I first connect the A5 pin or the GND pin. This only happens when the Router is plugged in (The router is turned off, of course). But when I unplug the router everything works just fine. I tried touching the two wires via some metal parts (including the housing of the router) and never got this weird behavior.
Has anyone else encountered this issue or does anyone maybe know what I’m missing out in this setup?

Arduino is just connected to a PC via USB, no other sources of power or ground

abdset, 2020-05-24:
Make sure you connect GND wire to the bit of the router and the A5 pin wire to the probing piece.

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?