grbl/grbl #1617: does GRBl upport G68 and G69?

Slixxor, 2020-03-22:
Looking at adding some easy auto straightening features to GRBL using a probe. I wondering if it supports G68 and G69 commands?

Rob2029, 2020-03-22:
I don't know, but you can try to send the command, with a finger on the E-Stop just in case, if the response is "ok" then it supports the command

terjeio, 2020-03-22:
G68 and G69 is not supported and neither is [G10 L2 with R word](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2), however some senders supports rotation.

Slixxor, 2020-03-22:
> 
> 
> G68 and G69 is not supported and neither is [G10 L2 with R word](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2), however some senders supports rotation.

Thanks for that. Do you know which senders support it?

closed 2020-03-22 by Slixxor

terjeio, 2020-03-23:
At least [GRBL-Plotter](https://github.com/svenhb/GRBL-Plotter) claims to do so. Never tested it though.