grbl/grbl #1619: XY zero function not working

ProblematicBachelor, 2020-03-27:
First, thank you in advance. I put my CNC together just today. And everything went smoothly until I came to set the XY axis using the tool in the corner. It was a grey colour and wasn't working. It is the same with the next three icons that are on the left of it. Any help is much appreciated, I am new so I am not offended if I have made a huge error. It is a bangood CNC so a cheap one. And I have looked and found no solution.

abdset, 2020-05-24:
Your post is missing so much informations, please provide us with more info so we try to help.
- What software are you using for your machine ?
- Is your machine based on GRBL board (Like arduino boards) ?
- When you say it is the same with next 3 icons, what icons (What they are supose to do) ?
- What is your CNC model and what OS are you using (Windows, Linux or MAC) ?

JEnoka1991, 2020-08-28:
This is the same problem i am having with the same router i'd say. It's a 3018 cnc. work and machine xyz co-ordinates are at zero. i upload the file to be machined. Move the spindle to the desired spot on the corner of my work piece, and then when i go to click the zero XY and the zero Z buttons, they dont work and still have the same co-ordinates as the machine co-ordinates. Also been looking for a way to make the jogging continuous, as when i have it in continuous mode, it only moves in increments of 0.333 
Any help would be greatly appreciated! I have a mac running grbl candle on an auduino board.