grbl/grbl #1620: Homing Works Fine.  When Running Gcode Z moves + and shuts everyhting down

gibsonsnet, 2020-03-31:
I am stuck.  Installed limit switches on Uno/CNC shield using GRBL 1.1 everything is great until I run my gcode.  Z goes up to the and stops the code.  Any help would be appreciated

**GRBL Settings**

CNCjs 1.9.22 [Grbl]
Connected to COM3 with a baud rate of 115200
Grbl 1.1h ['$' for help]
client> $$
[MSG:'$H'|'$X' to unlock]
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=500.000 (Homing locate feed rate, mm/min)
$25=500.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=12.700 (Homing switch pull-off distance, millimeters)
$30=1000 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=800.000 (X-axis travel resolution, step/mm)
$101=800.000 (Y-axis travel resolution, step/mm)
$102=800.000 (Z-axis travel resolution, step/mm)
$110=2000.000 (X-axis maximum rate, mm/min)
$111=2000.000 (Y-axis maximum rate, mm/min)
$112=2000.000 (Z-axis maximum rate, mm/min)
$120=2000.000 (X-axis acceleration, mm/sec^2)
$121=2000.000 (Y-axis acceleration, mm/sec^2)
$122=2000.000 (Z-axis acceleration, mm/sec^2)
$130=1000.000 (X-axis maximum travel, millimeters)
$131=1000.000 (Y-axis maximum travel, millimeters)
$132=1000.000 (Z-axis maximum travel, millimeters)

**Video of Homing**
https://www.youtube.com/watch?v=oTM8PvZnObk

**Video of Gcode**
https://www.youtube.com/watch?v=WuyFfq0b-WQ

**Gcode** (Generated from Fusion 360)

(1001)
(T1  D=3.17 CR=0 TAPER=30deg - ZMIN=-1 - chamfer mill)
G90 G94
G17
G21
G28 G91 Z0
G90

(Engrave2)
T1 M6
S5000 M3
G54
G0 X10.656 Y49.408
Z15
Z5
G1 Z0 F1000
X11.339 Y49.985 Z-1
X11.775 Y52.558
[... 4445 lines cut ...]
X167.876 Y50.65
Y50.626
X167.88 Y50.498
X167.881 Y50.478
X167.889 Y50.35
X167.89 Y50.33
X167.902 Y50.202
X167.904 Y50.183
X167.92 Y50.054
X167.922 Y50.04
X167.93 Y49.985
X168.151 Y49.797 Z-0.674
X168.42 Y49.573 Z-0.287
X168.622 Y49.408 Z0
X168.42 Y49.573 Z-0.287
X168.151 Y49.797 Z-0.674
X167.93 Y49.985 Z-1
X164.9
G0 Z15
G28 G91 Z0
G90
G28 G91 X0 Y0
G90
M5
M30

PicEngraver, 2020-04-05:
Grbl does not accept the tool change command 'M6'.  Edit out this line: 'T1 M6'.

PicEngraver, 2020-04-08:
Did removing "M6" solve your issue?  Feedback appreciated.

langwadt, 2020-04-08:
I suspect it is the "G28 G91 Z0" it moves the Z to 0 which hits the switch unless grbl is changed to zeros the axis after the pull off distance

In fusion360 you can untick the "safe retracts" in the post processor, just remember to raise the tool off the work before running the program

gibsonsnet, 2020-04-08:
Removing the G28 line did the trick thanks

closed 2020-04-08 by gibsonsnet