grbl/grbl #1622: x and y axis moves together for single axis command

aswin-thambi, 2020-04-15:
Hi,
 When i operate my machine,both the axis are moving together with a single axis command.
the machine move diagonally with any X or Y axis command.
can you help me to fix it?

Norwegian-Gadgetman, 2020-04-15:
That sounds like a HW issue.
Please verify that the cabling is correct.
That the two steppers doesn't share any cables.
That the cables are not terminated together at the board.
That the board if configurable doesn't 'slave' any stepper drivers together.
(This is sometimes done when you have 2 motors on one axis and one driver isn't capable of delivering a high enough current.)

aswin-thambi, 2020-04-15:
Sir,
I checked my connections several times,
Iam using arduino uno with tb6560 stepper drivers, and tried both grbl
controler and ugs.
The axises are moves perpendicular when i click the diagonal movement option

Norwegian-Gadgetman, 2020-04-15:
Please upload a drawing of your system, with every wire connecting the Arduino, the drivers and the motors.
Also make certain to include every Ground and power wire.

langwadt, 2020-04-15:
SW is build for or should be build for CoreXY?

aswin-thambi, 2020-04-15:
What may be the reasons sir,is there any cod eerror?

abdset, 2020-05-24:
If your machine has only one motor for each axis then make sure you don't have any jumper pluged next to the motors power connector of the arduino uno. They are used to share same signal of one axis to drive two motors that moves that axis.