grbl/grbl #1623: Enhancement: Multiple enable pins

goosenphil, 2020-04-17:
While grbl is nearly maxing out the mega328 for 3-axis machines in pin usage, it should be considered that some pins can be re-purposed for 2-axis users such as laser machines.

I've built a laser machine based on the DRV8825, it seems that because both drivers are enabled, if a motor isn't moving it gets the max amount of current driven into it and heats up. So it would be ideal to have multiple enable pins.

I would like to make a patch, but I need pointers where to look in the source code.

Looking at the this function:
https://github.com/grbl/grbl/blob/3ce1a9d637f05e28462a36cb8b166386aab94afc/grbl/stepper.c#L183

I can see that it set's PB0, which is Pin 8 of the Arduino when it asks the steppers to wake up.
Is there a way to detect within this function if only one motor is being used for the motion?

Norwegian-Gadgetman, 2020-04-17:
The current going to the one motor that isn't moving is to hold it in position.

It may be possible to adjust the current level 'on the fly' to take it down to a holding current that won't overheat your steppers too soon if you use an I2C-enabled stepper driver, such as some of Panucatt's 'Bigfoot' driversBut that would require either a rewrite of GRBL and a different shield, or a completely different controller board and possibly an updated version of the Smoothieboard SW.

But unless your motors gets too hot to touch, there really isn't a reason to do anything.

Trygve

goosenphil, 2020-04-17:
> The current going to the one motor that isn't moving is to hold it in position. It may be possible to adjust the current level 'on the fly' to take it down to a holding current that won't overheat [...]

What if I don't need a hold current? A laser is a very low torque load, the only issue I can see is if microstepping is used some steps could potentially be lost.

langwadt, 2020-04-17:
the current in the steppers are nearly the same whether it is moving or not, until you run out of supply voltage. Steppers are designed to run hot, I don't see any reason to disable a stepper unless the machine is idle and even that is like to cause missed steps

goosenphil, 2020-04-19:
@langwadt I see your point, thanks for clarifying my misunderstandings, I'll close the issue.

closed 2020-04-19 by goosenphil