grbl/grbl #1625: How do I INVERT the Y and Z Axis of my GRBL CNC?

ValentineKilla, 2020-04-24:


olvll, 2020-04-28:
I'm not sure how you generated that parameter list but it seems wrong (or non-standard to say the least). Look at the list at [Grbl v1.1 Configuration](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration) and note the differences.

In a normal Grbl setup you configure the directions of your axis with the [$3](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask) parameter.

For example $3=6 would invert both Y and Z, simple as that.

If you are using a version of Grbl that was hacked/adapted to work with a specific board or machine I guess it is possible that whoever did that, left the "direction port invert" parameter out and you will have to take it up with them.

Either way you could always just switch the wires from your stepper motors. That will make them turn the other way too.

ValentineKilla, 2020-04-29:
Hey Guys, 

Thanks for the feedback. 
I managed to find another post with a similar issue which matched my settings. 

Apparently each axis has a value assigned to them. 32 , 64 & 128. 
Depending on which one you wish to invert. You simple change $6= accordingly. 

For example.  
$6=32 
$6=64 
$6=128 

If you wish to invert more than one axis, simply add the numbers together. 
32 + 64 = 96 ($6=96)

Now as you can see my current value is 192 (no idea why!?) 
That means 64 and 128 have been inverted. 

Sooo I took a wild guess and decided to see what happens if I set the value to zero. 
$6=0 

IT WORKED!!
This removed the inversion on the two troubled axis Y & Z. 

Thanks for reading and I hope this helps anyone else in future. :)

Take care! 

Kind Regards,
VK

closed 2020-04-29 by ValentineKilla