grbl/grbl #1627: GRBL- 0.9 and UGS-Active state:UNKNOWN

eood3xm, 2020-04-30:
Hello,
I have worked successfully with Grbl-0.9 for Arduino Uno and UGS (for very simple cnc machine), but in recent days, I have a problem:
 After “connect”, appears message-“Error while processing response<Idle>” and  “Active state: UNKNOWN”. I reinstalled Arduino (the last version 1.8.12), and UGS - 2.0 Classic (Nightly build), but unfortunately problem was the same.
Work Position and Machine Position do not work. There are pulses to stepper and they have run. I have loaded a short program and it was running (without of control of real position). “Jog Mode” send pulses to stepper also, and they have run. “Visualize” also do not work (only static picture of contour).
I made attempt with UGS - 2.0 Platform (Beta) – without successes. 
Can you give me any advice in this situation?
Best regards

abdset, 2020-05-24:
Try this in UGS:

**For connection type:**
- Go to Tools then Options.
- Click on UGS icon (the last one on the right).
- Select Sender Options tab.
- Set Connection driver to JSerialComm.

**For board selection:**
- Make sure your are disconnected from the machine (Check in the icon below File menu).
- Next to that icon Select GRBL in Firmware option.

This assume you are using USB cable to connect to the PC, not a bluetooth connection.
If you are using bluetooth, you will have two COM ports dedicated to that connection, then you need to switch to the other COM port and try.

This error can also happend when your GRBL flash is corrupted, you will need to reflash the arduino with grbl.

To make it simple: This error means your connection to the machine is a succes but the software is not receiving the expected data from the arduino.

The arduino IDE software has nothing to do with this error but this make me thinking:
Make sure you close arduino IDE app before you connect the machine to UGS.

eood3xm, 2020-05-25:
Thank you very much, abdset. I have succeeded to reflash Grbl 1.1h and UGS Platform, and now 
everything is correct.
Thank you again.

closed 2020-05-25 by eood3xm