grbl/grbl #1629: Remeber what caused limit switch activation

amirm, 2020-05-03:
When a limit switch is activated, GRBL could remeber which direction it was going, so it can "get out" of it.

I have a machine with end stops on both ends of all axes.
When I accidentally hit the limit switch which is not the home switch, I have a hard time getting out of it.
I must reset GRBL, disable alarms, move a little, some 3 or 4 times.

Below is a suggestion, 
I had to use underlines to ident pseudo code. 
To see it better, copy to notepad and change all __ (double underlines) to double spaces.

**In some point of the program:**
enum LIMIT_REASON
{
__LIMIT_NOT_ACTIVATED,
__LIMIT_ACTIVATED_WHEN_GOING_PLUS,
__LIMIT_ACTIVATED_WHEN_GOING_MINUS,
__LIMIT_ACTIVATED_SNAFU
};
byte X_limit_reason = LIMIT_NOT_ACTIVATED;
byte Y_limit_reason = LIMIT_NOT_ACTIVATED;
byte Z_limit_reason = LIMIT_NOT_ACTIVATED;

**When moving, but not in HOMING:**
if (cfg21 && X_limit_activated) // Hard limits enabled AND limit switch activated
{
__if (x_going_plus())
____X_limit_reason = LIMIT_ACTIVATED_WHEN_GOING_PLUS; // Remember!
__else
____if (x_going_minus())
______X_limit_reason = LIMIT_ACTIVATED_WHEN_GOING_MINUS; // Remember!
____else
______X_limit_reason = LIMIT_ACTIVATED_SNAFU; // Watta... Limit switch activated, but axis was not moving!!!
__halt_movement();
}

**When HOMING**
if (cfg21 && X_limit_activated) // Hard limits enabled AND limit switch activated
{
__if (x_going_plus() && (X_limit_reason == LIMIT_ACTIVATED_WHEN_GOING_MINUS))
____ignore_limit_switch; // GRBL is going forth, and last limit activation was when going back

__if (x_going_minus() && (X_limit_reason == LIMIT_ACTIVATED_WHEN_GOING_PLUS))
____ignore_limit_switch; // GRBL is going back, and last limit activation was when going forth
}
else
{
__// Limit switch is not activated, forget why it was activated last
__X_limit_reason = LIMIT_NOT_ACTIVATED; // Forget!
}

**Same code, in one block**
if (cfg21) // Hard limits enabled
{
__if (homing())
__{
____// Homing...
____if (X_limit_activated) // limit switch activated
____{
______if (x_going_plus() && (X_limit_reason == LIMIT_ACTIVATED_WHEN_GOING_MINUS))
________ignore_limit_switch; // GRBL is going forth, and last limit activation was when going back
______if (x_going_minus() && (X_limit_reason == LIMIT_ACTIVATED_WHEN_GOING_PLUS))
________ignore_limit_switch; // GRBL is going back, and last limit activation was when going forth
____}
____else
____{
______// Limit switch is not activated, forget why it was activated last
______X_limit_reason = LIMIT_NOT_ACTIVATED; // Forget!
____}

____if (Y_limit_activated) // limit switch activated
____...

____if (Z_limit_activated) // limit switch activated
____...
__}
__else
__{
____// NOT Homing..
____if (X_limit_activated) // limit switch activated
____{
______if (x_going_plus())
________X_limit_reason = LIMIT_ACTIVATED_WHEN_GOING_PLUS; // Remember!
______else
________if (x_going_minus())
__________X_limit_reason = LIMIT_ACTIVATED_WHEN_GOING_MINUS; // Remember!
________else
__________X_limit_reason = LIMIT_ACTIVATED_SNAFU; // Watta... Limit switch activated, but axis was not moving!!!
______halt_movement();
____}

____if (Y_limit_activated) // limit switch activated
____...

____if (Z_limit_activated) // limit switch activated
____...
__}
} // cfg21 Hard limits enabled

jahnj0584, 2020-05-03:
Even easier is to just set up soft limits 1mm away from where the hard
limit triggers.

abdset, 2020-05-24:
As far as i know, GRBL does not get out of the actual closed limit switch (Maybe this needs to be improuved), in my machine i need to manualy move the axis from closed limit switch then i do reset followed by homing, but you can also do reset then move it by software then when it detects its out of limit switch the alarm with show up again then do second reset followed by homming and you are good to go.

Here is simpliest way to read this:

Manual method:
- Manualy move the axis out of limit switch.
- Press Reset button on controller or software.
- Do homing.

Software method:
- Press Reset button on controller or software.
- Move the axis out of limit switch using the software jog buttons (An alarm will show up when done).
- Press Reset again on controller or software.
- Do homming.