grbl/grbl #1646: Grbl 1.1 with CNC shield 3.0 - Spindle not responding

Quirogaj, 2020-07-11:
Hello everyone,

I just started a CNC project, like most people im new to this!

Overview of what im using so far:

3 axis (X, Y and Z.) DIY
CNC Shield 3.0
Arduino Uno
Grbl 1.1
G code Sender

I managed to flash Grbl 1.1 into the Arduino Uno, Install and move all three axis, they work perfect, g code sender can run the nc files im uploading.

Im just stuck with the spindle....

Here is what i have tried:

1) On config.h enable and desable #define SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED.
2) Conecting spindle to SpnEn and SpnDir.
3) Input M3 S200 and M4 S200 and just M3 or M4.

No result with this attemps....

But the best i have accomplish is the following:

4) Conecting the pins to Z+ and input "M3 S200"
       The spindle does not move at 200 rpm but i get a little sound when i activate it and the sound stops by disabling it with M5.

Please I need some adivce...

Thank you for your time.

PicEngraver, 2020-07-12:
Send M3 S1000 or M3 S12000.  It depends on what value you have grbl setting $30= set to.

Quirogaj, 2020-07-12:
I had $30= 255, now i moved it to $30= 20000 but still no luck.

>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 20000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 50.000    (X-axis travel resolution, step/mm)
$101 = 50.000    (Y-axis travel resolution, step/mm)
$102 = 50.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)

I tried M3 S200 M3 S500 M3 S12000 and up to 20000 rpm, i noticed the sound of the motor keeps going up but no movement shown.

Attached are the 3 pictures that show how everything is conected:

[image]
[image]
[image]

If i disconect it and plug a 12 V power supplu directly to the spindle, run absolutely normal, in case the question comes up.

terjeio, 2020-07-12:
You cannot drive a DC motor from a stepper driver output or directly from the controller pins. You need to add a relay or a DC motor driver circuit to make it work. Eg. a [L298N based driver](https://www.google.com/search?q=l298n+dc+motor+driver) can be used if you need to control the speed.

Quirogaj, 2020-07-12:
@terjeio Thank you so much for your time, @PicEngraver  you too.

I already watched a video on how to use the L298N base driver you mentioned, ill purchase one and conect the PWM signal from the Arduino during this week, ill let you know how it goes.

Just to make sure, the PWM signal is the one making the little sound on the DC motor right? and it is making the sound because there is not enough Voltaje provided to turn right?.

PicEngraver, 2020-07-12:
A PWM signal is chopped, not a pure DC voltage, so yes, likely causing the motor coils to vibrate slightly, making a sound.

Quirogaj, 2020-07-13:
@PicEngraver and @terjeio i wired the L298N with the Z+ as the PWM signal.

It worked just perfect, from 2000 to 20000 rpm. i still dont know how to verify the rpm, but that would be another topic further this week.

Thanks for the help.

Quirogaj, 2020-07-13:
Once the problem is solved, should i close this issue?

Let me know your comments.

terjeio, 2020-07-13:
> i wired the L298N with the Z+ as the PWM signal.

So you remapped the PWM output to that pin? The default PWM output is to Digital Pin 11.

> i still dont know how to verify the rpm

You need a tachometer for that.

> Once the problem is solved, should i close this issue?

Yes.

Hosseinnemati, 2021-08-21:
Hi friends, I started cnc shield with Arduino, I was able to control the axes through a software but I could not control the spindle through the program I installed, Also, when I send the $ and $$ commands, I do not receive a reply even though I set it to 115200.
Grbl 1.1h ['$' for help]
This is also the version of the program.