grbl/grbl #1647: Spindle control with relay?

hitchhiker85, 2020-07-19:
hi!

i got problems with spindle control.
i put a relay on pin d11.
it stays everytime high.

whats wrong?
what must change to work with on/off via m3/m5?

thanks

PicEngraver, 2020-07-20:
Try:
(For Off)
S0 M5
(For On)
S1000 M3

terjeio, 2020-07-20:
Is the relay coil is connected directly to the pin (usually a bad idea) or via a buffer? Either way do not forget a [snubber diode](https://en.wikipedia.org/wiki/Snubber) across the coil or risk killing the MCU or the buffer.

hitchhiker85, 2020-07-20:
its a relay shield.
with a transistor,led,diode....

not only a relay!

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?