grbl/grbl #1650: 1 axis, 2 motors going opposite ways and doing weird stuff

ItsFl0ris, 2020-07-29:
Okay, so i'm new to GRBL but i have rebuild the MIRANDA CNC and i'm struggeling with grbl.

As you can see with the miranda cnc design, the Y axis has 2 motors. I configured the A driver to replicate the movement of the Y motor. But they are going opposite ways and i've already burnd a driver trying to get this working.

Anyone that knows a fix for this?
Sorry for my bad english btw!

langwadt, 2020-07-29:
swap the two wires on one of the coils on the stepper going the wrong direction

ItsFl0ris, 2020-08-01:
isn't there a way to do this with software or something else or do i really have to swap the cable's itself because its a real pain in the ass to make those little connectors