grbl/grbl #1651: math.h compliance

kimstik, 2020-07-31:
As per GRBL not used double but float, in a way to respect math.h declarations "funcf" should be used instead of "func"
atan2	-> atan2f
floor	-> floorf
round	-> roundf
sin	-> sinf
cos	-> cosf
sqrt	-> sqrtf
fabs	-> fabsf

terjeio, 2020-07-31:
It does not matter for 8-bit Arduinos as double precision [is the same](https://www.arduino.cc/reference/en/language/variables/data-types/double/) as single precision float.

langwadt, 2020-07-31:
but it does matter if porting to another platform, especially something with a single precision FPU

kimstik, 2020-07-31:
Exactly.
It allow me to fit GRBL to about 15k bytes of ARM thumb code.

On other hand what is good reason to violate math.h declarations? Because it works?

terjeio, 2020-07-31:
> On other hand what is good reason to violate math.h declarations?

IIRC parameter values gets promoted to double before the call and the result truncated to float on ARM. So it could be that the result will be more precise at the cost of slower execution speed. Even slower if a FPU is present and it is not used?

> Because it works?

It does but I do not consider it a good reason. I have switched to the float equivalents in my port to avoid the overhead..

kimstik, 2022-07-04:
avr-libc in the right direction
https://github.com/avrdudes/avr-libc/commit/e08bbd25e260804090e3e55e40bbcac3dc8f4abd