grbl/grbl #1652: GRBL does not zero machine coordinates after homing

Slixxor, 2020-07-31:
I would expect that after a homing cycle is completed that the achine coordinates would zero.

I have checked there are no offsets and have reset the coordinate system using $RST=#

It still shows negative numbers in the machine coords.

How can these be zero or is there something wrong?

jahnj0584, 2020-07-31:
It accounts for the back off distance of your switch and depending what
corner is home you'll have negatives. Top right corner would result in X+Y
being negative.

Slixxor, 2020-08-02:
> 
> It accounts for the back off distance of your switch and depending what corner is home you'll have negatives. Top right corner would result in X+Y being negative.
> […](#)

Hi,

This is much more than that. if it backed off 357mm on X to get bak to home it'll set the axis to -357mm on machine coords.

Is there any ideas why?