grbl/grbl #1656: Axis not homing properly

jkwilborn, 2020-08-18:
I have installed optical limit switches at both ends of each axis on my 3018 CNC.  They all function properly if manually triggered and will also trigger when you attempt to jog over the limits on any axis at either end.  I have good software and hardware skills so hopefully I can respond with the proper information.  I also have a 4 channel digital scope if we need that.

The Z axis appears to homes properly and gives me an idea on how the homing works, or at least an assumption.  When the X axis is homing, it will go until it triggers the limit, backs off the switch and gives me a 9 error.  It does not slowly re-engage the limit switch a second time as the Z axis does, the fine adjustment part, I guess.  Modifying either or both $3 and/or $2 to 2 causes an ALARM:8, but the basic movements appear to be the same as far as I can tell by 'eye.'  This is also confusing as this is supposed to affect only the Y axis...?

Since the wiring has been checked, I've attempted to extend the pull-off for alarm 8 and reduce for 9 as the documentation suggests, but to no avail.  I understand the meaning of what $2 does, electrically, but am not clear on it's impact on the actual machine operation. I assume it's to allow the hardware to play nice together.   So do I configure for the 9 error or the 8 alarm and debug from there? 

This is difficult to 'google' for as there are millions of "how to's" on using and debugging limit switches but none on this issue.  So after a couple of weeks of beating my head against the wall on this I thought I'd try and ask for help.  Since the software has been pretty well hammered on, and it all seems to work, hardware wise, I'm assuming it's configuration, which follows. (**original parameters in parentheses.**)  This is a [woodpecker cnc board](https://www.amazon.com/Engraving-Machine-Control-Engraver-Controller/dp/B07B3RFNJ7/ref=sr_1_5?dchild=1&keywords=woodpecker+control&qid=1597768203&sr=8-5).

Grbl 1.1f ['$' for help]
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok
$0=10
$1=25
$2=0 (2)
$3=0 (2)
$4=0
$5=0
$6=0
$10=16
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0 (I believe this had Y inverted and was 2, as it would home to the front)
$24=10.000
$25=500.000
$26=250
$27=0.250
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=1000.000
$111=1000.000
$112=600.000
$120=400.000
$121=50.000
$122=50.000
$130=200.000
$131=200.000
$132=200.000
ok

Best I can determine, and I may be all wet, is that it's going the wrong direction after the initial limit contact...  Don't know how to modify this or if it's the right approach.

Difficult to figure out all of the inversion bits and how they interact, so... I reset all of the directional variable to non invert and saw which way the axis were moving when I issue the $H.  The Z and the X axis move in the positive direction, up for Z and to the right for X.  I never get to see the Y axis move since the error occurs during the X axis homing.  I had thought I'd read that after the Z axis cleared the work, both X and Y would move together. That does not occur, it appears to be sequential and do Z, X and then Y, but I see no movement in either Y directions.  

Thanks for all of your time and effort that benefits the rest of us.  Any suggestions would be appreciated.

PicEngraver, 2020-08-19:
On my Woodpecker v3.4 board, the X and Y switch inputs are marked reversed.  Try swapping your X and Y inputs to see if it fixes your problem.

My setting are currently:
$0=30
$1=254
$2=0
$3=2
$4=0
$5=0
$6=0
$10=0
$11=0.005
$12=0.002
$13=1
$20=1
$21=0
$22=1
$23=3
$24=50.000
$25=2000.000
$26=250
$27=1.000
$30=1001
$31=0
$32=0
$100=810.000
$101=810.000
$102=810.000
$110=2000.000
$111=2000.000
$112=2000.000
$120=50.000
$121=50.000
$122=50.000
$130=171.450
$131=90.000
$132=40.000
I currently use this machine as a software test machine and frequently change settings, so mine may not siuit your setup.  My homing is to the bottom left of the table.
Hope this helps.
John Champlain
www.picengrave.com

jkwilborn, 2020-08-20:
John, thanks so much for the note and information.  I went to only a single
limit switch on each axis as there was less hardware.  I also followed the
GRBL guide about it seeking the max X, Y and Z initially and expecting
limits there.  I set it up as they suggested.  I removed all the inverting
bits.  Set the Y bit for 'step pulse.' Set it to home and adjusted the
reverse bits to ensure everything was going the right way for a home
operation.  After a lot of tinkering, I got it to 'home.'  Since I have a
CNC 3018 with a woodpecker grbl controller I thought I'd go ahead and post
the current configuration.  I'm doing this so others might have more luck
with this issue.  Debugging this is tough... :(

Thanks for bringing up the problem with the incorrect screen graphic,
swapping the X and Z axis limit switches on the Woodpecker version 3.4
card.  I was aware of that from previous research, but it's a good to know
as I only found it in one other place.

$0=10

$1=25

$2=2

$3=0

$4=0

$5=0

$6=0

$10=19

$11=0.010

$12=0.002

$13=0

$20=0

$21=1

$22=1

$23=2

$24=50.000

$25=500.000

$26=250

$27=1.000

$30=1000

$31=0

$32=0

$100=800.000

$101=800.000

$102=800.000

$110=1000.000

$111=1000.000

$112=600.000

$120=30.000

$121=30.000

$122=30.000

$130=180.000

$131=320.000

$132=50.000

I think I can safely enable the soft limits now that the home function
appears working.... I also looked at the code and found where I can do
separate homes for each axis.  However, for some reason, it's not
compiling, on the Arduino IDE. When I have time, I'll look at it more
closely. This configuration only has ensured that the limit switches are
working. Again, thanks so much for putting up with us idiots... :)

Jack :)

PicEngraver, 2020-08-20:
Thanks for the feedback.