grbl/grbl #1660: Set stepper motor as spindle in grbl v1.1

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?
im using 2 stepper motors for x,y axis , solenoid for z axis , stepper motor as spindle
im really in rush for solving this problem
please

jahnj0584, 2020-08-28:
Stop posting this everywhere or i will have to ban you immediately.

ygjfggfgfj, 2020-08-28:
im very sorry  jahnj0584
could you help me in this issue of stepper motor?

ygjfggfgfj, 2020-08-28:
im very sorry
please dont think im annoying person, but i have project and my teacher
changed the hardware "spindle" from dc motor to stepper, and i dont know
how to control the speed of it using PWM that GRBL provides. my project
should finish in this week.
could you help me, or you know anyone could help me.
finally accept my appologie.

Norwegian-Gadgetman, 2020-08-29:
GRBL doesn't support using a stepper as the spindle.

And trying to use the 'free' Z axis driver won't work because a spindle is expected to run at constant speed, and that will be impossible to code for in the G_code it interprets.

How long ago is it since your teacher changed your spindle?
And what was his reasons?

Trygve

ygjfggfgfj, 2020-08-29:
My teacher changed the spindle because he doesn't have a Dc spindle and because the purpose of project is to control syrange for conductive ink which need amount of accuracy

Norwegian-Gadgetman, 2020-08-29:
Hi!

This sounds more like a pinter. I would really suggest that you look at Arduino FirmWare for 3D printers, as they DO have a '4th axis' controlled by a stepper.

ygjfggfgfj, 2020-08-29:
Thanks a lot
I will try to find software for 3D printer