grbl/grbl #1661: Relationship between spindle speed and S command

Jackflash79, 2020-08-30:
Hello,
I have a 3018-like CNC with a grbl 1.1 board and a Chinese 12000rpm rated spindle.
I used a digital tachometer to measure the spindle speed in response to the S command, tied to different values of $30 and $31. My goal was to get some sort of exact value.
The problem is that the relationship between the setpoint value S and the measured spindle speed is not linear. It is rather logarithmic.
I was wondering if there was a way, or some planned development, for grbl to handle this.

[image]

langwadt, 2020-08-30:
you are looking at the wrong repository for 1.1,  a piecewise fit is already in 1.1h , https://github.com/gnea/grbl

https://github.com/gnea/grbl/blob/master/doc/script/fit_nonlinear_spindle.py

Jackflash79, 2020-08-30:
Thank you for pointing me to the right place ;-)