grbl/grbl #1662: GRBL ERROR 20

poolrod, 2020-09-08:
Not sure if I'm using this site correctly, but I'm giving it a shot. My programs I tried to run has errors 20, 24 and 33,, is there another G-code I can put in where the errors are and keep my design intent?. Thank you for the help.
[test error errors.txt](https://github.com/grbl/grbl/files/5189253/test.error.errors.txt)

PicEngraver, 2020-09-09:
What program generated your gcode file?  It appears to not have been setup correctly for use with grbl.

poolrod, 2020-09-09:
I'm trying to use Mastercam and was told a generic Fanuc post would work but might need tweaked a little.

PicEngraver, 2020-09-09:
Perhaps this Mastercam forum thread from 2019 will help you.
https://www.emastercam.com/forums/topic/97978-mastercam-91-w-grbl/

PicEngraver, 2020-09-09:
Sorry, can't really help with Mastercam.  I saw where you already posted on their forum.

poolrod, 2020-09-09:
Thank you for trying John. Have a great day.