grbl/grbl #1665: Strange receive of "ok" from marlin v1.1

Mikron007, 2020-09-12:
Hi
I made myself a laser engraver and it stops working at random position, but comunication stays open.
From LightBurn software console I noticed strange receive from controller. I'm using Arduino Uno GRBL shield.
I changed the USB cable for shorter one (about 0.5 m shielded cable) and that didn't solve my problem.
I added picture of my problem.
[image]

Mikron007, 2020-09-14:
Now i'm trying Universal Gcode Sender - UGS and I getting similar results. Laser still stops in the middle of the job.
[image]
[image]

langwadt, 2020-09-14:
settings?

Mikron007, 2020-09-14:
> settings?

My grbl settings are:
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=255
$31=0
$32=1
$100=100.000
$101=100.000
$102=250.000
$110=6000.000
$111=6000.000
$112=6000.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

terjeio, 2020-09-15:
Some Arduino clones has a lousy USB <> UART chip? There are lost characters so worth investigating?

Mikron007, 2020-09-15:
> Some Arduino clones has a lousy USB <> UART chip? There are lost characters so worth investigating?

Tnx. I have one arduino from Keyestudio. I will try that one.

Mikron007, 2020-09-27:
It seems that changing the arduino uno board solved the problem.
Tnx.

closed 2020-09-27 by Mikron007