grbl/grbl #1666: use arduino serial monitor to send grbl ascii  "0x92" real time command

Hubert880411, 2020-09-15:
I use this method:First, in config.h you need to change the following line for the new character

#define CMD_SAFETY_DOOR 0x92 ---> #define CMD_SAFETY_DOOR '&'
then in serial.c, case CMD_FEED_OVR_COARSE_MINUS: system_set_exec_motion_override_flag(EXEC_FEED_OVR_COARSE_MINUS); break;
just move this up about 3 lines to where you see the line starting with
case CMD_FEED_HOLD:
And I type '&' on Arduino serial monitor when g code is running.But it appears error1 on monitor.Please tell me how to trigger feed override command.

MeJasonT, 2020-09-15:
it will have to be in the gcode not an MDI command.
Its is easier just to open the door suerly

MeJasonT, 2020-09-15:
Its actually a nifty little feature to have, the industrial machines dont retract they leave the spindle in situ and ramp down the rpm to 300 for example.  Some machines lock the doors shut and you have to use a feed hold button to release the doors.  ideally you want to be as close as possible to a pro machine for two reasons.  To learn how to use machines properly - then its not a big shock when you end up using one eventually and its safer to stick to the tried and tested method.  why do you want to send the door control in code - it doesnt really make sense.

Hubert880411, 2020-09-15:
> Its actually a nifty little feature to have, the industrial machines dont retract they leave the spindle in situ and ramp down the rpm to 300 for example. Some machines lock the doors shut and you [...]

Thanks for replying me again Jason.I want to use adxl345(Accelerometer) and grbl to achieve vibration feedback.But I don't know how to do if I don't use Arduino sketch.

terjeio, 2020-09-15:
> Please tell me how to trigger feed override command.

Holding down <Alt> and typing 0146 works from PuTTY on Windows. Don't know if this also works from the Arduino monitor...

0146 = 0x92