grbl/grbl #1668: HOW I Can run vexta 5 phase stepper motor

jafarsofian, 2020-09-25:
hello world 
anyone can help me run my vexta stepper motor contains 5 wires connection by pentagon wound i have a driver (b0655180) i attach some picture for stepper motor and driver**
[image]
**
[image]

saeed3hsani, 2022-11-08:
Hello 
I have same problem !
Did you find any solutions?

jlipavsky, 2022-11-09:
Is there an advantage to using this over regular steppers?