grbl/grbl #1672: grbl homing issue

Roelandjelmer, 2020-10-06:
Dear Experts,

I can't seem to get the homing cycle to run on the Gcode Sender. I have no issue at all jogging the motors through the controller but when I try to run the homing cycle, nothing happens, the only thing it says is "home".
[image]
 All of my end switches work propperly (NC) and I use an filter. Everything is solderd
[image]
[image]

I'm using the following configuration:
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 2    (dir port invert mask:00000010)
$4 = 0    (step enable invert, bool)
$5 = 1    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 1    (soft limits, bool)
$21 = 1    (hard limits, bool)
$22 = 1    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 500.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 53.330    (x, step/mm)
$101 = 53.330    (y, step/mm)
$102 = 400.000    (z, step/mm)
$110 = 2500.000    (x max rate, mm/min)
$111 = 2500.000    (y max rate, mm/min)
$112 = 1000.000    (z max rate, mm/min)
$120 = 50.000    (x accel, mm/sec^2)
$121 = 25.000    (y accel, mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 600.000    (x max travel, mm)
$131 = 600.000    (y max travel, mm)
$132 = 600000.000    (z max travel, mm)

Hope someone can help me.
Thanks!

helLf1nGer, 2020-10-07:
I'm not an expert, but I would suggest upgrading firmware to grbl 1.1h. Also that serial output looks weird

Roelandjelmer, 2020-10-25:
Hello helLf1nGer, spending much time on trouble shooting, thinking that the firmware would be good. Tanks this solved the problem!

closed 2020-10-25 by Roelandjelmer