grbl/grbl #1673: Digital Spindle Control

G4GXO, 2020-10-08:
I am aware that this has been raised before but I'm not clear as to whether a solution is available yet - (apologies if is and if so I'd be grateful if someone could direct me towards it). 

I have an old Bungard CNC which I use to prototype PCBs. The controller is HPGL driven from a DOS PC based program. I'm in the process of building a new machine controller to make the CNC hardware compatible with modern software using a G-Code interface. I have the X-Y-Z steppers working nicely and I can drive the KaVo HF spindle manually from a mixed C/Assembler program running in a dsPIC33E processor which drives a home made MOSFET inverter. Spindle speed is from 5000 to 60,000RPM.

It would be desirable to drive the spindle speed from GRBL, placing speed control within milling and drilling software. I could simply use the filtered Uno PWM to generate a speed voltage which could be read by the dsPIC33E ADC but it would be better to be able to pass exact speed as a numerical value, ideally 16bits.

Has anyone succeeded in modifying GRBL to include a simple serial interface to pass speed in this way rather than as a PWM duty cycle?

I also need to consider how to control the compressed air chuck via GRBL, I wonder if this feature also exists - I don't see an output for this on the Uno, perhaps I need to consider a different Arduino platform?

Many thanks,

Ron Taylor

G4GXO, 2020-10-11:
Don't act upon this, I've decided to build the functionality into my dsPIC33 spindle driver using the Capture/Compare module. By using this approach compatability will be provided with any future releases of GRBL where the PWM spindle output of the Arduino will hopefully remain as a standard speed control interface.

My plan is to measure duty cycle and scale max RPM within the dsPIC33 accordingly. The results will be smoothed by an exponential moving average filter and a spot of hysteresis applied to prevent the spindle RPM from hunting. The resulting RPM resolution does not need to be high and the 1024 range of the PWM is more than enough for my purposes.

If there is interest in this I can post my driver hardware and software design once complete and fully tested.

Regards

Ron

Bongaloa, 2023-11-27:
Hello Ron,

Yes please. Just bought myself a bungard and want to replace the brain with something more modern.

Lieuwe

Bongaloa, 2023-12-02:
> Hi Lieuwe,
>
> My apologies for the delay in responding but I’m wrapped up in a DSP
> project at present. I’ve started posting details of the PWM spindle driver
> that I built for my old Bungard. The speed is set manually using a rotary
> encoder, I have not found a need yet to drive it via G-Code. I used GRBL as
> the basis for my X-Y-Z control. This runs on an Arduino 328 connected to
> the host PC via USB. I fitted a screw terminal hat to the Arduino to
> simplify connections to the stepper drivers. I used three 4A 9V-42V TB6600
> stepper drivers purchased through Amazon. Power for the steppers comes from
> a 24V 6.5A PSU, a second 12V 3A PSU supplies the PWM driver board with a
> third 5V PSU supplying the Arduino and PWM processor board.
>
> The general arrangement is described
>
> On the PC I use KiCAD to design a board and generate Excellon and Gerber
> files, FlatCAM to generate milling and drilling G-Code data and Universal
> G-Code Sender to send the data to the controller to drive the machine. It
> all works very well indeed!
>
> I’ve started uploading spindle project data to GitHub. Allow me a day or
> two to find everything and upload it.
>
> Regards
>
> Ron Taylor

Hello Ron,

Thanks for your reply.

I intend to use the bungard both for milling and laser cutting/engraving.

Also got my hands on a kavo sf 4005 and want to see if I can get that
working with the original controls.

Will read through the details and keep an eye on github.

Again, thanks for this usefull information.

Regards,
Lieuwe Banga