grbl/grbl #1674: Power Control To Steppers , revisit

getgray, 2020-10-12:
I have an X-Y only application where I need to be able to power down the stepper's hold current while not in use.  But I need them to stay put for a few seconds at least after the last command.  My "tool" will not suffer from lost microsteps, if I even do microstepping.  My duty cycle is low for the motors, and GCODE will come in "spurts" with some seconds of delay between them, and somtime's many minutes between the next commands.  Having the motors on all the time just isn't practical for my application.  But, I need the "brain" to be alive and well, waiting for the next move at any moment, without and bootup delays other than the enable.  If I send a reset as a "wake", I'm likely to lose GCODE commands being send immediately after the wake.  I can lose microsteps, I cannot lose GCODE motion commands.  

I'm "controlling" GRBL from an upstream device that is feeding the GCode via a human interface.  I can develop an idle time timer there and when triggered, send commands to GRBL to disable.  I gather I can use the "Sleep/$SLP" command to turn off the motors.  That's great, but I need to not reboot to turn them back on.

Aside, why Ctrl-X for reset?  I'd be grateful for a text command (vs. CTRL F) to "wake" (reset) the device so I don't have to send "Ctrl-X" as some terminal devices won't allow the character (Universal Gcode Sender for one).  Is a total "Sleep" necessary to turn the step motor power on and off?  I read a previous issue where it was suggested to bend some pin so one could tie in a switch on the enable pin.  Is that necessary if you aren't using the CNC shield board, going to a larger driver instead?

Surely there's a better way to signal enable/disable?  Particularly with larger drivers like a Gecko 201 for example.  I can control pins, relays, whatever from the upstream control device that's sendign the GCODE, but I need a way to communicate via GRBL to shut down the motors.  Ideally without a reboot/reset, although for the latter, a simpler $ command to reset would be appreciated.

A timeout value for $1 was great I thought at first.  But then I discovered that 254 ms was the max delay and useless for my purposes.  I would like to hold things steady for inertia reasons for a few seconds so I don't lose whole steps.  But after that it can go "limp", pending the next move.  I was hoping to be able to set that hold till delay to 10,000ms (10s) instead.

Maybe I'm missing some obvious solution or misunderstanding a detail, grateful for any assistance in a workaround/solution.

langwadt, 2020-10-12:
if you can live with it not responding during the delay, just change line 224 in stepper.c

PicEngraver, 2020-10-12:
"I have an X-Y only application where I need to be able to power down the stepper's hold current "
Why?  Why can the steppers not stay powered?  A clear answer might help you to the right solution.

getgray, 2020-10-12:
> "I have an X-Y only application where I need to be able to power down the stepper's hold current "
> Why? Why can the steppers not stay powered? A clear answer might help you to the right solution.
They coudl of course, btu I'm going to be using big, high current steppers that may sit idle for 90% of the time.  Locking the steps isn't an issue, sitting there pulling a lot of current for possibly hours on end is an issue, to me.  If for no other reason that wasting the power supply and electricity.  But the user won't like it either.  I can, (and lilky will if I resolve this) leave the hold current on for a few minutes after the most recent user pause. but turn off if it goes idle for a while.  This is not a CNC router where this behavior would not make sense, it does it job then quits.  My application has intermittent, active use.

getgray, 2020-10-12:
> if you can live with it not responding during the delay, just change line 224 in stepper.c

I'm not following.  Grateful for some elaboration.  I see the st_wake_up function, I don't see how changing it's name would help.  Maybe you meant something else.

langwadt, 2020-10-12:
line 224 is the delay before disabling the steppers

getgray, 2020-10-12:
> line 224 is the delay before disabling the steppers

Line 224 on my stepper.c (grbl-1.1h.20190825)  is:
void st_wake_up()

langwadt, 2020-10-12:
if you are using 1.1 why are you posting in the repository for 0.9?

https://github.com/gnea/grbl/blob/master/grbl/stepper.c line 262

PicEngraver, 2020-10-12:
Connect the coolant pin (A3) and use M8/M9 to toggle the stepper controller enable pins from your program.  A G4Px delay before toggling pin A3 would provide your time delay before disabling?  Just a thought.

getgray, 2020-10-12:
> 
> 
> if you are using 1.1 why are you posting in the repository for 0.9?
> 
> https://github.com/gnea/grbl/blob/master/grbl/stepper.c line 262

Because I'm a git dummy, sorry.  

I assumed I couldn't just go changing that delay value to something relatively long (like 5 minutes) without screwing up interrupts somewhere.

getgray, 2020-10-12:
> 
> 
> Connect the coolant pin (A3) and use M8/M9 to toggle the stepper controller enable pins from your program. A G4Px delay before toggling pin A3 would provide your time delay before disabling? Just a thought.

I like the sounds of that suggestion.  Allow me a dummy question.  triggering enable/disable will release the motor, right?  Hold current ($1=255) will stop until the enable pin is flipped back to enable?  Without causing a controller reset?

Getting OT but not sure if the CNC shield board will allow me to test that, that's what the other issue was discussing where someone said a pin had to be bent wasn't sure what they meant.

PicEngraver, 2020-10-12:
Getting a little beyond my pay grade now, but using the coolant pin to control the stepper enable function should have no effect on the $1=255 setting - leave it at 255 I think.  I suggest you look at documentation for your stepper drivers as to what happens to stepper power when toggling the enable pin.  No controller reset should occur with toggling coolant pin else it would be worthless to use.

Easy to test pin A3 with a voltmeter.  No bending of pins necessary I think?  If your CNC shield board does not have pin A3 available (my gShield boards do not) then connect straight to the Uno board itself. ???

getgray, 2020-10-12:
> Getting a little beyond my pay grade now, but using the coolant pin to control the stepper enable function should have no effect on the $1=255 setting - leave it at 255 I think. I suggest you look [...]

Got it, thank you.  The "spare" coolant control should be a a viable IO path.  My controlling device can issue the Gcode and determine the ideal idle period.  Agree, then I can leave $1=255 so it's on hold untill/unless told to go to sleep via enable control.  This assuming that the enable (disable) pin shuts current to the motor.  That, I'm fuzzy on.  Investigating now.

terjeio, 2020-10-13:
In my 32-bit port I have changed the "go idle" delay to register a callback to a function that disables the motors when the delay  expires. If motion is restarted during the delay period the callback is cancelled. A viable option for 8-bit too?

getgray, 2020-10-13:
> 
> 
> In my 32-bit port I have changed the "go idle" delay to register a callback to a function that disables the motors when the delay expires. If motion is restarted during the delay period the callback is cancelled. A viable option for 8-bit too?

That sounds like an excellent solution, albeit a little over my skill level to implement.  Any pointers?  I'm running GRBL on a Arduino Uno FWIW.  I'm open to upgrading that hardware if that solves it.  Can you elaborate or share details?

terjeio, 2020-10-13:
> Any pointers? 

You can find my port [here](https://github.com/terjeio/grblHAL). The setting value for the delay is still limited to max 255 ms, but I see no reason for not lifting this limit - it is a matter of changing the delay variable from 8 to 16 bit.

> I'm running GRBL on a Arduino Uno FWIW.

Then a [Nucleo-64 board](https://www.st.com/en/evaluation-tools/nucleo-f411re.html) could be used, I have drivers for STM32F4xx processors - support for the STM32F446 is ready to be commited to github. Note that these processors runs at 3.3V - you have ensure the shield is usable with that.

Top of the range of supported processors is the iMRXT1062 @ 600MHZ found on Teensy 4.x boards.

> Can you elaborate or share details?

More details are available on the github page,

getgray, 2020-10-13:
terjeio:  That looks impressive, but implementation is out of my skillset.  I'd need more handholding/how-to's to implement than are evident to me.  Alas, with Arduino there are enough dummy guides and getting that with GRBL was pretty straightforward.  I do like the sounds of your port though.