grbl/grbl #1677: Spindle maximum acceleration setting

jiquera, 2020-10-22:
This has probably be requested before but would be possible to add a spindle maximum acceleration setting just like with the axis motors? It would make life so much easier on managing current draw from spindles and make much more spindles compatible with specific setups.

I realize I can do this manually by sending a gcode sequence but this seems like a setting that would be useful for a lot of people and doesn't seem particularly hard to implement (at least at first glance).

Cheers

jiq

jahnj0584, 2020-10-22:
That's the job of the VFD or power supply to do.

jiquera, 2020-10-23:
Well yeah, but you could also argue that the spindle rotation direction and the axis directions could be handled by switching manually wires but there are settings for that. Additionally I suppose that a significant part of the target audience (home cnc users) has psus and vfds that don't have this feature (none of the budget cncs I found do this) and need little workarounds like I mentioned above.

terjeio, 2020-10-23:
Some [grblHAL drivers](https://github.com/terjeio/grblHAL/tree/master/drivers) has support for ramped spindle RPM, but I have to admit I have not tested this in a while. You may get an idea of how to implement this for 8-bit processors from my code - but I am not sure if the required resources (a spare timer?) are available for doing that.

jiquera, 2020-10-23:
I got a bit confused with the grbl main page so I posted it twice (old and new wiki): https://github.com/gnea/grbl/issues/944

I haven't looked at any grbl source code yet but since the goal here is current limiting, I think the accuracy doesn't need to be great. Which is why I think this could be solved in software without a feedback loop. I didn't take available resources into account yet, though... I suppose the typical worst case solution (a busy loop) would cause problems elsewhere in the system, right :p