grbl/grbl #1679: GRBL port to NXP imxRT1020

Moffy, 2020-10-31:
I have mostly ported GRBL to the RT1020EVK board, which has an Arduino shield socket. The serial comms is working fine (using CDC usb) and I am about to test the stepper routine. I might have a few questions about coding, when, if I get stuck.
Thanks for a great project.

terjeio, 2020-10-31:
Then you may get some ideas from my code for the [iMXRT1060](https://github.com/terjeio/grblHAL/tree/master/drivers/IMXRT1062) on a [Teensy 4 board](https://www.pjrc.com/store/teensy41.html), there are support for SD card and ethernet as well in that.

Moffy, 2020-11-01:
Thank you, I have had a look at your code previously and it is impressive. Unfortunately I am using the MCUXpresso development software which has a different set of dependencies, and thereby makes your code for me a little harder to understand. I might look further at how you implement the serial connection(I think you use usb CDC with a sysint? Which is how I am attempting it) as I am having a few issues there.

terjeio, 2020-11-01:
> Unfortunately I am using the MCUXpresso development software

Then take a look at the [LPC1769 driver](https://github.com/terjeio/grblHAL/tree/master/drivers/LPC1769).

> I might look further at how you implement the serial connection(I think you use usb CDC with a sysint? Which is how I am attempting it) as I am having a few issues there.

Don't do it like that? Polling on the systick event is an Arduino thing. I hook code directly into the USB stack for LPC1769, cannot remember now if it was by implementing a weakly defined function (`usbBufferInput`). I do it in a similar way for the STM32 drivers - this by adding it as "USER CODE".

I do not know what your goal with this port is, anyway - why not write a driver for grblHAL instead? ;-)

Moffy, 2020-11-01:
You must excuse me but I am a bit more of a dabler than proficient programmer. It all started because I have the RT1020EVK board and I had an idea of adapting GRBL for it as it already has the Arduino shield pins for it, nicely spaced out. It was only after I was well into the little project that I ran across yours and some other code. Writing a "driver" sounds like a good idea, but I am not at present sure how to proceed with that. Your ability and understanding is well above mine as I was mostly modifying framework examples and merging them together with modifications.
If you have any advice about how to write the driver I would be willing to listen.
Thank you.

terjeio, 2020-11-01:
> Writing a "driver" sounds like a good idea, but I am not at present sure how to proceed with that.

Fork grblHAL to your github account and I'll help out. Copying the LPC1769 driver as a starting point is perhaps best, the iMRXT driver has Arduino dependencies that should be avoided. Setting up an environment for development is a bit tricky. Use the approach I use for the [Arduino drivers](https://github.com/terjeio/grblHAL/wiki/Using-Eclipse-for-Arduino-development), this will make it a lot easier to commit to your fork. When your driver is complete you can make a pull request against the grblHAL master if you want that.

After you have forked grblHAL open an issue at your repository and we can continue the discussion there.

Moffy, 2020-11-01:
I have forked your repository and opened an issue as requested. The url is: https://github.com/Moffy/grblHAL
Thanks again, look forward to your advice.