grbl/grbl #1680: Z axis homing problem

ssombra, 2020-11-03:
I have a DIY laser engraver which I built using an old 3D printer frame and parts, a UNO with a CNC Shield and a 5.5W diode láser. I flashed it with GRBL 1.1 and I am using LaserGrbl. Once I had it working, I added endstops switches for X and Y, only 1 per axis, as min switches with the left bottom side as origin (3D printing background :)) and changed the homing sequence for X and Y only, again everything was working good. I then decided to also add an endstops switch for min Z, as my setup included also Z axis motors and I think that it would be a good idea to be able to home Z at the height with the laser is focused on the bed and then be able to move up depending on the height of the material to engrave.
I installed the switch, connecting it to “Z min”, and changed the homing sequence to be: Z, then X and Y.
The problem I have is that when I press the home button, the z axis starts moving but it doesn’t stop even when the switch is pressed. I checked for continuity and the switch is working good (I.e, continuity when pressed). Am I missing something? My laser PWM is connected to Z max, does that have something to do? Should I change something?

KevinBaileyCrum, 2020-11-17:
are your switches normally open or closed?
did you change the limit pin invert? https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#5----limit-pins-invert-boolean

JanuszKr, 2023-02-25:
The solution for the CNC shield v.3 and Arduino Uno is very simple: just plug the Z endstop switch to the pins labeled "spin enable", while the laser is pluged to Z axis endstop pins. I did it and it solved the problem.

JanuszKr, 2023-02-25:
In addition, for homing, the $21 must be set to 0.